7Ji/orangepi5-archlinuxarm #1: sd card wont boot

DanimanX, 2023-02-23:
I tested image ArchLinuxARM-aarch64-OrangePi5-20230217_134642.img.xz and when I power up orange pi there is no green light so there is no boot (I tested previously debian image on that sd card and it's work).

DanimanX, 2023-02-23:
Thanks for the quick response, I was able as you mention to install SPI Flash but your image only boot via USB with a tf adapter, If I want to boot via microsd slot it's stuck on the red led.

7Ji, 2023-02-24:
I could boot it in sd slot without problem, just tested again to verify. Do you have a USB-UART adapter to check what's going on when doing a SD boot?

7Ji, 2023-02-25:
I just downloaded the image from Github release (to confirm if the upload itself is corrupted) and wrote it to my cheapest microSD card on hand, and it boots without any problem from that card:
```
nomad7ji@ms5 ~ > ssh alarm@alarm.lan
alarm@alarm.lan's password: 
[alarm@alarm ~]$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
mtdblock0    31:0    0    16M  0 disk 
mmcblk1     179:0    0  58.2G  0 disk 
├─mmcblk1p1 179:1    0    99M  0 part /boot
└─mmcblk1p2 179:2    0   1.9G  0 part /
zram0       254:0    0     0B  0 disk 
nvme0n1     259:0    0 232.9G  0 disk 
├─nvme0n1p1 259:1    0   100M  0 part 
├─nvme0n1p2 259:2    0     8G  0 part 
└─nvme0n1p3 259:3    0 224.8G  0 part 
[alarm@alarm ~]$ cat /etc/os-release 
NAME="Arch Linux ARM"
PRETTY_NAME="Arch Linux ARM"
ID=archarm
ID_LIKE=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinuxarm.org/"
DOCUMENTATION_URL="https://archlinuxarm.org/wiki"
SUPPORT_URL="https://archlinuxarm.org/forum"
BUG_REPORT_URL="https://github.com/archlinuxarm/PKGBUILDs/issues"
```

Unless you have more logs from UART with some more info to debug containing the logs from bootloader (u-boot), it's hard to understand what's going wrong on your side, I'll close the issue for now

closed 2023-02-25 by 7Ji