7Ji/orangepi5-archlinuxarm #2: Opi-5b wifi + spi not detected/working properly

DaveWK, 2023-07-14:
Hi,
Thanks for putting this together; a cleanly made build repo for an OrangePi build is a refreshing change. 
I don't seem to be able to get the onboard wifi for the orange pi 5b to work, however. 

I plugged in an external USB wifi, which does work, as well as the wired ethernet works, so it seems to be something lower level in the startup. I see in the logs:
```[    7.925078] rockchip-spi feb20000.spi: register misc device rkspi-dev2
[    7.926540] spi-nor spi5.0: unrecognized JEDEC id bytes: ff ff ff ff ff ff
[    7.926614] spi-nor: probe of spi5.0 failed with error -2
[    7.927776] rk_gmac-dwmac fe1c0000.ethernet: IRQ eth_lpi not found
[    7.927964] rk_gmac-dwmac fe1c0000.ethernet: Looking up phy-supply from device tree
[    7.927976] rk_gmac-dwmac fe1c0000.ethernet: Looking up phy-supply property in node /ethernet@fe1c0000 failed
[    7.927989] rk_gmac-dwmac fe1c0000.ethernet: no regulator found
[    7.928040] rk_gmac-dwmac fe1c0000.ethernet: clock input or output? (output).
[    7.928098] rk_gmac-dwmac fe1c0000.ethernet: TX delay(0x42).
[    7.928145] rk_gmac-dwmac fe1c0000.ethernet: Can not read property: rx_delay.
[    7.928203] rk_gmac-dwmac fe1c0000.ethernet: set rx_delay to 0xffffffff
[    7.928266] rk_gmac-dwmac fe1c0000.ethernet: integrated PHY? (no).
[    7.928319] rk_gmac-dwmac fe1c0000.ethernet: cannot get clock mac_clk_rx
[    7.928375] rk_gmac-dwmac fe1c0000.ethernet: cannot get clock mac_clk_tx
[    7.928443] rk_gmac-dwmac fe1c0000.ethernet: cannot get clock clk_mac_speed
[    7.928714] rk_gmac-dwmac fe1c0000.ethernet: init for RGMII_RXID
[    7.928893] rk_gmac-dwmac fe1c0000.ethernet: User ID: 0x30, Synopsys ID: 0x51
[    7.928953] rk_gmac-dwmac fe1c0000.ethernet:         DWMAC4/5
[    7.928997] rk_gmac-dwmac fe1c0000.ethernet: DMA HW capability register supported
[    7.929057] rk_gmac-dwmac fe1c0000.ethernet: RX Checksum Offload Engine supported
[    7.929116] rk_gmac-dwmac fe1c0000.ethernet: TX Checksum insertion supported
[    7.929173] rk_gmac-dwmac fe1c0000.ethernet: Wake-Up On Lan supported
[    7.929254] rk_gmac-dwmac fe1c0000.ethernet: TSO supported
[    7.929300] rk_gmac-dwmac fe1c0000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    7.929365] rk_gmac-dwmac fe1c0000.ethernet: Enabled Flow TC (entries=2)
[    7.929419] rk_gmac-dwmac fe1c0000.ethernet: TSO feature enabled
[    7.929469] rk_gmac-dwmac fe1c0000.ethernet: Using 32 bits DMA width```

But then the wifi never shows up when using `networkctl` .

7Ji, 2023-07-14:
This might sound stupid but are you using the DTB for 5B instead of 5?

5B has eMMC instead of SPI flash, so with 5B's DTB there shouldn't be log about SPI NOR.

And could you share the result of `lspci`? I don't have a 5B myself but judging from the difference (lacking m.2 slot than 5) the WiFi module should be internally connected via PCI-E bus and you should at least see the module.

DaveWK, 2023-07-14:
ope, you're right -- needed the 5b dtb, but still doesn't seem to be working right. Before when I rtan `lspci` it wqas blank, but now I get:
```[alarm@alarm ~]$ lspci 
0004:40:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0004:41:00.0 Network controller: Broadcom Inc. and subsidiaries Device 449d (rev 02)
```

in my dmesg I see:

``` 13.142022] [dhd] PCI_PROBE:  bus 41, slot 0,vendor 14E4, device 449D(good PCI location)
[   13.145066] [dhd] dhdpcie_init: found adapter info 'DHD generic adapter'
[   13.148189] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 3, size 139264
[   13.148197] [dhd] succeed to alloc static buf
[   13.148198] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 4, size 0
[   13.148243] pcieh 0004:41:00.0: enabling device (0000 -> 0002)
[   13.159190] [dhd] Disable CTO
[   13.159838] [dhd] ******** Perform FLR ********
[   13.180822] [BT_RFKILL]: bt shut off power
[   13.238515] [dhd] read_config: reg=0x88 read val=0xb080
[   13.248630] [dhd] read_config: reg=0x88 read val=0x8080
[   13.254370] [dhd] ******** FLR Succedeed ********
[   13.257198] [dhd] DHD: dongle ram size is set to 1310720(orig 1310720) at 0x170000
[   13.259932] [dhd] dhdpcie_bar1_window_switch_enab: bar1_switch_enab=0 ramstart=0x170000 ramend=0x2affff bar1_size=0x0
[   13.262701] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 7, size 35072
[   13.265445] [dhd] dhd_conf_set_chiprev : devid=0x449d, chip=0xaae8, chiprev=2, svid=0x14e4, ssid=0xaae8
[   13.268198] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 0, size 1776
[   13.272355] [dhd] dhd_pktid_map_init:2458: pktid_audit init succeeded 1025
[   13.275160] [dhd] dhd_pktid_map_init:2458: pktid_audit init succeeded 4097
[   13.278172] [dhd] dhd_pktid_map_init:2458: pktid_audit init succeeded 36865
[   13.282447] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 5, size 65536
[   13.286785] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 19, size 65712
[   13.296682] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 20, size 262144
[   13.303608] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 22, size 65536
[   13.307166] [dhd] dhd_attach(): thread:dhd_watchdog_thread:199 started
[   13.310601] [dhd] dhd_deferred_work_init: work queue initialized
[   13.313689] [dhd] dhd_tcpack_suppress_set: TCP ACK Suppress mode 0 -> mode 3
[   13.316783] [dhd] dhd_tcpack_suppress_set: TCPACK_INFO_MAXNUM=10, TCPDATA_INFO_MAXNUM=10
[   13.319905] [dhd] dhd_cpumasks_init CPU masks primary(big)=0xf0 secondary(little)=0xe
[   13.325876] [dhd] dhd_get_memdump_info: MEMDUMP ENABLED = 3
[   13.329155] [dhd] dhdpcie_bus_attach: making DHD_BUS_DOWN
[   13.332424] [dhd] dhdpcie_init: rc_dev from dev->bus->self (1d87:3588) is 000000000a99966e
[   13.336131] [dhd] dhdpcie_init: rc_ep_aspm_cap: 1 rc_ep_l1ss_cap: 1
[   13.339690] [dhd] dhdpcie_request_irq: INTx enabled
[   13.342857] [dhd] dhdpcie_bar1_window_switch_enab: bar1_switch_enab=0 ramstart=0x170000 ramend=0x2affff bar1_size=0x400000
[   13.346046] [dhd] dhd_bus_download_firmware: firmware path=/lib/firmware/ap6275p/fw_bcmdhd.bin, nvram path=/lib/firmware/ap6275p/nvram.txt
[   13.349149] [dhd] dhdpcie_dump_resource: BAR0(VA): 0x000000004b3d671d, BAR0(PA): 0x00000000f4200000, SIZE: 32768
[   13.351978] [dhd] dhdpcie_dump_resource: BAR1(VA): 0x000000009c4a710f, BAR1(PA): 0x00000000f4400000, SIZE: 4194304
[   13.354722] [dhd] dhd_conf_set_path_params : Final fw_path=/lib/firmware/ap6275p/fw_bcm43752a2_pcie_ag.bin
[   13.357472] [dhd] dhd_conf_set_path_params : Final nv_path=/lib/firmware/ap6275p/nvram_AP6275P.txt
[   13.360202] [dhd] dhd_conf_set_path_params : Final clm_path=/lib/firmware/ap6275p/clm_bcm43752a2_pcie_ag.blob
[   13.362934] [dhd] dhd_conf_set_path_params : Final conf_path=/lib/firmware/ap6275p/config.txt
[   13.369840] [dhd] dhd_get_download_buffer: Open image file failed /lib/firmware/ap6275p/config.txt
[   13.372465] [dhd] dhd_conf_read_config : Ignore config file /lib/firmware/ap6275p/config.txt
[   13.375027] [dhd] d2h_intr_method -> PCIE_INTX(0); d2h_intr_control -> D2H_INTMASK(0)
[   13.377690] [dhd] dhdpcie_download_code_file: dhd_tcm_test_enable 0
[   13.380245] [dhd] dhdpcie_download_code_file: download firmware /lib/firmware/ap6275p/fw_bcm43752a2_pcie_ag.bin
[   13.382812] [dhd] dhdpcie_download_code_file: Open firmware file failed /lib/firmware/ap6275p/fw_bcm43752a2_pcie_ag.bin
[   13.385670] [dhd] _dhdpcie_download_firmware:4813 dongle image file download failed
[   13.388744] [dhd] dhd_bus_start: failed to download firmware /lib/firmware/ap6275p/fw_bcm43752a2_pcie_ag.bin
[   13.391846] [dhd] dhdpcie_init: dhd_bus_start() failed
[   13.394499] [dhd] dhdpcie_bus_remove_prep: making DHD_BUS_DOWN
[   13.397130] [dhd] dhd_detach: making dhdpub up FALSE
[   13.399737] [dhd] dhdpcie_oob_intr_unregister: irq is not registered
[   13.402522] [dhd] dhd_detach(): thread:dhd_watchdog_thread:199 wait for terminate
[   13.405154] [dhd] dhd_detach(): thread:dhd_watchdog_thread:199 terminated OK
[   13.425297] [dhd] dhd_rtt_deinit: ENTER
[   13.432052] [dhd] dhd_rtt_deinit: EXIT, err = 0
[   13.435799] [dhd] dhd_tcpack_suppress_set: TCP ACK Suppress mode 3 -> mode 0
[   13.439411] [dhd] dhd_tcpack_suppress_set: TCPACK_INFO_MAXNUM=10, TCPDATA_INFO_MAXNUM=10
[   13.442347] [dhd] dhd_flow_rings not initialized!
[   13.449517] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 7, size 0
[... 24 lines cut ...]
[... cut ...]
[  314.410721] rk_gmac-dwmac fe1c0000.ethernet end1: stmmac_open: Cannot attach to PHY (error: -19)
```

but still no adapter under `nmcli` for the wifi:
```[root@alarm ~]# nmcli 
lo: connected (externally) to lo
        "lo"
        loopback (unknown), 00:00:00:00:00:00, sw, mtu 65536
        inet4 127.0.0.1/8
        inet6 ::1/128
        route6 ::1/128 metric 256

end1: disconnected
        "end1"
        1 connection available
        ethernet (rk_gmac-dwmac), EE:C6:57:7A:DF:60, hw, mtu 1500
```

7Ji, 2023-07-15:
https://github.com/DHDAXCW/rockchip-rootfs-firmware/raw/main/fw_bcm43752a2_pcie_ag.bin

This link should be the needed firmware, you can download it to try

```
sudo mkdir -p /usr/lib/firmware/ap6275p
sudo wget https://github.com/DHDAXCW/rockchip-rootfs-firmware/raw/main/fw_bcm43752a2_pcie_ag.bin -O /usr/lib/firmware/ap6275p/fw_bcm43752a2_pcie_ag.bin
```

The following patch should add the firmware as a pre-loaded blob, but since I don't have a opi5b I want to wait for your test results about the firmware first.
```
From d33b3cf40efd3a5dc737447a8974231caa9f86e5 Mon Sep 17 00:00:00 2001
From: 7Ji <pugokushin@gmail.com>
Date: Sat, 15 Jul 2023 15:39:06 +0800
Subject: [PATCH] config: add blob for firmware of opi 5b's onboard m.2 Wi-Fi
 module

---
 config.sh | 12 ++++++++++++
 1 file changed, 12 insertions(+)

diff --git a/config.sh b/config.sh
index 4fb88a0..eb26876 100644
--- a/config.sh
+++ b/config.sh
@@ -10,4 +10,16 @@ name_distro+='-OrangePi5'
 release_note_packages+=(
   'linux-aarch64-orangepi5:[my AUR][AUR linux-aarch64-orangepi5]'
   'yay:[AUR][AUR yay]'
+)
+blob_paths+=(
+  '/usr/lib/firmware/ap6275p/fw_bcm43752a2_pcie_ag.bin'
+)
+blob_modes+=(
+  '0644'
+)
+blob_urls+=(
+  'https://github.com/DHDAXCW/rockchip-rootfs-firmware/raw/d771fb7efd8ffff68a8f62bd44ed41d7c414e39c/fw_bcm43752a2_pcie_ag.bin'
+)
+blob_sha256sums=(
+  'bfcdc3ecb5274745f3c3551abd0d9b11ede89b305837241364a055fefbf09de7'
 )
\ No newline at end of file
-- 
2.41.0

```

DaveWK, 2023-07-15:
I tried putting that `fw_bcm43752a2_pcie_ag.bin` in the location you specified (and chmod-ing it to `0644`) but it seems like it needs the file `nvram_AP6275P.txt`?

I am getting the error:
```
  13.494254] [dhd] dhd_bus_download_firmware: firmware path=/lib/firmware/ap6275p/fw_bcmdhd.bin, nvram path=/lib/firmware/ap6275p/nvram.txt
[   13.494257] [dhd] dhdpcie_dump_resource: BAR0(VA): 0x0000000020dad239, BAR0(PA): 0x00000000f4200000, SIZE: 32768
[   13.494258] [dhd] dhdpcie_dump_resource: BAR1(VA): 0x000000006ae5d507, BAR1(PA): 0x00000000f4400000, SIZE: 4194304
[   13.494265] [dhd] dhd_conf_set_path_params : Final fw_path=/lib/firmware/ap6275p/fw_bcm43752a2_pcie_ag.bin
[   13.494267] [dhd] dhd_conf_set_path_params : Final nv_path=/lib/firmware/ap6275p/nvram_AP6275P.txt
[   13.494268] [dhd] dhd_conf_set_path_params : Final clm_path=/lib/firmware/ap6275p/clm_bcm43752a2_pcie_ag.blob
[   13.494269] [dhd] dhd_conf_set_path_params : Final conf_path=/lib/firmware/ap6275p/config.txt
[   13.501053] [dhd] dhd_get_download_buffer: Open image file failed /lib/firmware/ap6275p/config.txt
[   13.521389] [dhd] dhd_conf_read_config : Ignore config file /lib/firmware/ap6275p/config.txt
[   13.521393] [dhd] d2h_intr_method -> PCIE_INTX(0); d2h_intr_control -> D2H_INTMASK(0)
[   13.521502] [dhd] dhdpcie_download_code_file: dhd_tcm_test_enable 0
[   13.535714] [dhd] dhdpcie_download_code_file: download firmware /lib/firmware/ap6275p/fw_bcm43752a2_pcie_ag.bin
[   13.538209] [dhd] dhd_os_open_image1: /lib/firmware/ap6275p/fw_bcm43752a2_pcie_ag.bin (857142 bytes) open success
[   13.627477] [dhd] dhd_get_download_buffer: Open image file failed /lib/firmware/ap6275p/nvram_AP6275P.txt
[   13.630041] [dhd] _dhdpcie_download_firmware:4848 dongle nvram file download failed
[   13.632518] [dhd] dhd_bus_start: failed to download firmware /lib/firmware/ap6275p/fw_bcm43752a2_pcie_ag.bin
[   13.635021] [dhd] dhdpcie_init: dhd_bus_start() failed
[   13.637524] [dhd] dhdpcie_bus_remove_prep: making DHD_BUS_DOWN
[   13.640066] [dhd] dhd_detach: making dhdpub up FALSE
[   13.642575] [dhd] dhdpcie_oob_intr_unregister: irq is not registered
[   13.645152] [dhd] dhd_detach(): thread:dhd_watchdog_thread:18f wait for terminate
[   13.647693] [dhd] dhd_detach(): thread:dhd_watchdog_thread:18f terminated OK
[   13.676519] [dhd] dhd_rtt_deinit: ENTER
[   13.679162] [dhd] dhd_rtt_deinit: EXIT, err = 0
[   13.681828] [dhd] dhd_tcpack_suppress_set: TCP ACK Suppress mode 3 -> mode 0
[   13.684697] [dhd] dhd_tcpack_suppress_set: TCPACK_INFO_MAXNUM=10, TCPDATA_INFO_MAXNUM=10
[   13.687581] [dhd] dhd_flow_rings not initialized!
[   13.694552] [dhd] STATIC-MSG) dhd_wlan_mem_prealloc : sectoin 7, size 0
[   13.697263] [dhd] dhdpcie_pci_probe: PCIe Enumeration failed
[   13.699831] [dhd] dhdpcie_pci_probe : mutex is released.
[   13.702436] [dhd] dhdpcie_bus_register: dhdpcie initialize failed.
[   13.705028] [dhd] dhd_wifi_platform_load_pcie: pcie_register_driver failed
[   13.707610] [dhd] unregister wifi platform drivers
[   13.710205] [dhd] wifi_platform_bus_enumerate device present 0
[   13.712783] [dhd] ======== Card detection to remove PCIE card! ========
[   13.715381] [dhd] ======== dhd_wlan_deinit_plat_data ========
[   13.717981] [dhd] dhd_wlan_deinit_gpio: gpio_free(WL_HOST_WAKE 0)
[   13.720600] [dhd] _dhd_module_init: Failed to load the driver, try cnt 0
[   13.723245] [dhd] _dhd_module_init: Failed to load driver max retry reached**
[   13.725854] [dhd] _dhd_module_init: Exit err=-1
```

7Ji, 2023-07-16:
All four of them are needed:
```
[   13.354722] [dhd] dhd_conf_set_path_params : Final fw_path=/lib/firmware/ap6275p/fw_bcm43752a2_pcie_ag.bin
[   13.357472] [dhd] dhd_conf_set_path_params : Final nv_path=/lib/firmware/ap6275p/nvram_AP6275P.txt
[   13.360202] [dhd] dhd_conf_set_path_params : Final clm_path=/lib/firmware/ap6275p/clm_bcm43752a2_pcie_ag.blob
[   13.362934] [dhd] dhd_conf_set_path_params : Final conf_path=/lib/firmware/ap6275p/config.txt
```

7Ji, 2023-07-17:
All of them can be found at orangepi's firmware repo:
https://github.com/orangepi-xunlong/firmware/tree/master/ap6275p

7Ji, 2023-07-21:
Hello, can you elaborate further on the test results with the firmware files available in opi's repo applied? I'll close this without adding them if I can't get test results as I don't want to include stuffs not used.

DaveWK, 2023-07-21:
ope, sorry for the delay -- after I added the `ap6275p` files it can now detect wifi networks, but doesn't seem to be able to actually connect to them; possibly I need another piece of the firmware?

7Ji, 2023-07-21:
>  it can now detect wifi networks, but doesn't seem to be able to actually connect to them

Can you see the device as a wirelsss client (not DHCP client) on your access point? The whole "connect to a wireless AP" concept consists of at least two stages: the client assosiated with AP, no layer 2 transport, then network configured, with layer 2 transport. I don't know your network manager configuration, but usually the two stages need to be handled by at least two parts of software, e.g. `wpa_supplicant` for connecting the wireless, then `systemd-networkd` for configuring the network. If you're using `NetworkManager` then it automatically does the 2 stages for you.

>  a way to disable the "debug logs" for other components

Just `grep` then? The kernel kmsg buffer does not record any module info, it's up to the module to prefix their message, the magic color is simply added by userspace dmesg before the first `:` for each line. A simple
```
dmesg -w | grep -F [dhd] | tee /tmp/logfile
```
should work just fine

DaveWK, 2023-07-24:
I'm using the serial connection, but here's what I get now:

```
[   30.214593] [dhd] dhd_sync_with_dongle: GET_REVINFO device 0x449d, vendor 0x14e4, chipnum 0xaae8
[   30.217107] [dhd] dhd_sync_with_dongle: RxBuf Post : 2048
[   30.221999] [dhd] dhd_preinit_ioctls: preinit_status IOVAR not supported, use legacy preinit
[   30.222039] [dhd] dhd_tcpack_suppress_set: TCP ACK Suppress mode 0 -> mode 3
[   30.222063] [dhd] dhd_tcpack_suppress_set: TCPACK_INFO_MAXNUM=10, TCPDATA_INFO_MAXNUM=10
[   30.223877] [dhd] dhd_legacy_preinit_ioctls: hostwake_oob enabled
[   30.225217] [dhd] dhd_legacy_preinit_ioctls: use firmware generated mac_address b8:13:32:73:46:d4
[   30.225308] [dhd] dhd_os_open_image1: /lib/firmware/ap6275p/clm_bcm43752a2_pcie_ag.blob (30993 bytes) open success
[   30.227944] [dhd] dhd_check_current_clm_data: ----- This FW is not included CLM data -----
[   30.297067] [dhd] dhd_check_current_clm_data: ----- This FW is included CLM data -----
[   30.308251] [dhd] Firmware up: op_mode=0x0005, MAC=b8:13:32:73:46:d4
[   30.338531] [dhd] dhd_legacy_preinit_ioctls: event_log_max_sets: 26 ret: 0
[   30.358250] [dhd] arp_enable:1 arp_ol:0
[   30.358353] [dhd] dhd_conf_add_pkt_filter : 142 0 0 77 0xffffffffffffffffffffffffffffff 0x2F6465766963652F77616B6575702F
[   30.382895] [dhd]   Driver: 101.10.361.20 (wlan=r892223-20220701-3)
[   30.382895] [dhd]   Firmware: wl0: Jul 12 2022 18:54:54 version 18.35.387.23.146 (g412cc5ec) FWID 01-93c53be6
[   30.382895] [dhd]   CLM: 9.9.12_SS (2021-10-12 17:03:28) 
[   30.385607] [dhd] dhd_pno_init: Support Android Location Service
[   30.385718] [dhd] dhd_rtt_init ENTRY
[   30.388409] [dhd] rtt_do_get_ioctl: failed to send getbuf proxd iovar (CMD ID : 1), status=-23
[   30.388529] [dhd] dhd_rtt_init : FTM is not supported
[   30.388575] [dhd] dhd_rtt_init EXIT, err = 0
[   30.430416] [dhd] [INIT] logset:8 is preserve/chatty
[   30.436077] [dhd] [INIT] logset:10 is preserve/chatty
[   30.481469] [dhd] dhd_ecounter_autoconfig Ecounter autoconfig in FW not supported
[   30.517028] [dhd] dhd_rx_frame: net device is NOT registered. drop event packet
[   30.517505] [dhd] dhd_conf_set_country : set country CN, revision 0
[   30.522042] [dhd] dhd_rx_frame: net device is NOT registered. drop event packet
[   30.523017] [dhd] dhd_conf_set_country : Country code: CN (CN/0)
[   30.538837] [dhd] dhd_bus_devreset: WLAN Power On Done
[   30.538937] [dhd] [wlan0] wl_android_wifi_on : Success
[   30.571002] [dhd] dhd_rx_frame: net device is NOT registered. drop event packet
[   30.583369] [dhd] dhd_bus_aspm_enable_dev: RC ENABLE , linkctrl_before: 0x30110000 linkctrl_after: 0x30110002
[   30.583586] [dhd] dhd_bus_aspm_enable_dev: EP already enabled  linkctrl: 0x10110142
[   30.584056] [dhd] [wlan0] dhd_open : Exit ret=0
[   30.584105] [dhd] [wlan0] dhd_pri_open : tx queue started
[   30.584193] [dhd] [wlan0] custom_xps_map_set : Done. mapping cpu
[   30.584227] [dhd] dhd_pri_open : mutex is released.
[   30.593822] [dhd] [wlan0] _dhd_set_mac_address : close dev for mac changing
[   30.594009] [dhd] [wlan0] dhd_pri_stop : tx queue stopped
[   30.594048] [dhd] [wlan0] dhd_stop : Enter
[   30.594076] [dhd] dhd_stop: ######### called for ifidx=0 #########
[   30.594107] [dhd] [wlan0] dhd_stop : skip chip reset.
[   30.594138] [dhd] [wlan0] dhd_stop : Exit
[   30.596629] [dhd] [wlan0] _dhd_set_mac_address : MACID 02:3a:6b:04:d6:9d is overwritten
[   30.596740] [dhd] dhd_pri_open : no mutex held
[   30.596773] [dhd] dhd_pri_open : set mutex lock
[   30.596801] [dhd] [wlan0] dhd_open : Primary net_device is already up
[   30.596847] [dhd] [wlan0] dhd_pri_open : tx queue started
[   30.596898] [dhd] [wlan0] custom_xps_map_set : Done. mapping cpu
[   30.596929] [dhd] dhd_pri_open : mutex is released.
[   30.597123] [dhd] [wlan0] _dhd_set_mac_address : notify mac changed done
[   30.700665] [dhd] [wlan0] wl_run_escan : LEGACY_SCAN sync ID: 3, bssidx: 0
[   33.367500] [dhd] [wlan0] dhd_pri_stop : tx queue stopped
[   33.367563] [dhd] [wlan0] dhd_stop : Enter
[   33.367646] [dhd] dhd_stop: ######### called for ifidx=0 #########
[   33.367672] [dhd] dhd_rtt_deinit: ENTER
[   33.367691] [dhd] dhd_rtt_deinit: EXIT, err = 0
[   33.375729] [dhd] CFGP2P-ERROR) wl_cfgp2p_disable_discovery :  do nothing, not initialized
[   33.378522] [dhd] dhd_stop: making dhdpub up FALSE
[   33.378602] [dhd] dhd_tcpack_suppress_set: TCP ACK Suppress mode 3 -> mode 0
[   33.378622] [dhd] dhd_tcpack_suppress_set: TCPACK_INFO_MAXNUM=10, TCPDATA_INFO_MAXNUM=10
[   33.378679] [dhd] [wlan0] wl_android_wifi_off : in g_wifi_on=1, on_failure=1
[   33.378700] [dhd] dhd_rtt_deinit: ENTER
[   33.378714] [dhd] dhd_rtt_deinit: EXIT, err = 0
[   33.378730] [dhd] dhd_bus_devreset: == Power OFF ==
[   33.378767] [dhd] dhd_bus_stop: making DHD_BUS_DOWN
[... 283 lines cut ...]
[   35.861311] [dhd] dhd_bus_aspm_enable_dev: EP already enabled  linkctrl: 0x10110142
[   35.861720] [dhd] [wlan0] dhd_open : Exit ret=0
[   35.861749] [dhd] [wlan0] dhd_pri_open : tx queue started
[   35.861805] [dhd] [wlan0] custom_xps_map_set : Done. mapping cpu
[   35.861825] [dhd] dhd_pri_open : mutex is released.
[   35.867742] [dhd] [wlan0] _dhd_set_mac_address : close dev for mac changing
[   35.868013] [dhd] [wlan0] dhd_pri_stop : tx queue stopped
[   35.868053] [dhd] [wlan0] dhd_stop : Enter
[   35.868082] [dhd] dhd_stop: ######### called for ifidx=0 #########
[   35.868114] [dhd] [wlan0] dhd_stop : skip chip reset.
[... cut ...]
[   35.870864] [dhd] dhd_pri_open : mutex is released.
[   35.871060] [dhd] [wlan0] _dhd_set_mac_address : notify mac changed done
[   35.932667] [dhd] [wlan0] wl_run_escan : LEGACY_SCAN sync ID: 4, bssidx: 0
[   38.781573] [dhd] [wlan0] wl_run_escan : LEGACY_SCAN sync ID: 5, bssidx: 0
[   42.464873] [dhd] [wlan0] wl_run_escan : LEGACY_SCAN sync ID: 6, bssidx: 0
[   47.467914] [dhd] [wlan0] wl_run_escan : LEGACY_SCAN sync ID: 7, bssidx: 0
[   54.470627] [dhd] [wlan0] wl_run_escan : LEGACY_SCAN sync ID: 8, bssidx: 0
```

7Ji, 2023-07-25:
There's three `_dhd_set_mac_address : close dev for mac changing`, the driver is constantly closing the netdev then re-open it to change mac address. You're using NetworkManager? Turn off wifi.scan-rand-mac-addres.

And I can see the country code is set to CN, this is hardcoded in the driver for certain chips. What's your country? And I doubt being set to CN means the wireless chip cannot connect to your AP.

https://github.com/orangepi-xunlong/linux-orangepi/blob/eb1c681e5184e51d8ce1f351559d149d17f48b57/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/dhd_config.c#L4882

DaveWK, 2023-07-27:
After setting:
```
[device]
wifi.scan-rand-mac-address=no
```

It's now connecting to my wifi properly -- thanks!

closed 2023-07-28 by 7Ji

7Ji, 2023-07-28:
I just created a package with all the firmware on opi's repo included: https://aur.archlinux.org/packages/linux-firmware-orangepi

You can use yay to install that and uninstall `linux-firmware` to save space. The package will also later be pre-installed in the images available on this repo, and will be installable for users with old images via `pacman` directly from https://github.com/7Ji/archrepo

DaveWK, 2023-07-29:
Thanks! There seems to be several versions of the firmware floating around in various git repos, not sure which one is the newest/most reliable.

Can start a new thread, but 2 other drivers you might want to consider adding into the image:
the rockchip mpp drivers   -- https://aur.archlinux.org/packages/mpp-git 

the new panfrost-css drivers (bleeding edge, but it seems like they're adding support for the mali g610):
https://gitlab.freedesktop.org/panfrost/mesa/-/commits/panfrost/v10-wip/
and either "pan-css" or "panthor" (Not sure but "panthor" may be the new name of "pan-css"?)

https://gitlab.freedesktop.org/bbrezillon/linux/-/tree/pancsf

https://gitlab.freedesktop.org/bbrezillon/linux/-/tree/panthor?ref_type=heads

7Ji, 2023-07-30:
I'm keeping the image as minimal as possible, with only necessary components available to make the image bootable and connectable to the network, things like media playback, video de/encoding, graphics, desktop environment are not something I think everyone needs, and will certainly add up to the space taken.

The image will still be kept as minimal as it was. But i may add some prebuilt packages to my repo, and make the repo pre-configured for new images. That way, anyone wantinh those packages can just `pacman -Syu`

And is it a good idea to mix the panfrost driver and the vendor kernel? Usually you want the mainline kernel to use panfrost, and if you're using vendor kernel then you should just use the proprietary ARM driver. You can check that `panfrost` is not compiled at all from OrangePi's config (I didn't change their config at all for my image). `panfrost` is in the mainline tree for a long, long time, and if the vendor doesn't use it, it means they're using the ARM proprietary driver.

From experience, I am using `panfrost` and `lima` on some of my other ARM devices, but all of them are running mainline kernel (6.1 lts and 6.4). Their performance is not as good as vendor kernel + proprietary ARM driver. I don't see the point of vendor kernel + panfrost, just for performance penalty, not considering if that combination works at all, if your kernel tree is already miles away from the mainline.

```
nomad7ji@opi ~> zcat /proc/config.gz | grep PANFROST
# CONFIG_DRM_PANFROST is not set
```