7Ji/orangepi5-archlinuxarm #5: Does GPU / 3D Acceleration Work?

Integral-Tech, 2023-08-01:
Does GPU / 3D Acceleration work?
If it does NOT work, how to make GPU / 3D Acceleration work?

7Ji, 2023-08-01:
Sorry, I've never even connected my opi5 to any display. I just got opi5 for its performance to function as a native AArch64 build server for my various projects. And since it's under active heavy duty I don't want to get it offline to tinker with GPU stuffs. So you'll have to figure that out by yourself.

That being said, I don't think there's anything stopping the GPU acceleration from working. The kernel in this image was built with opi's unmodified official kernel config, and it has thousands of lines of patch I added to make GCC12 work with it. It was even built natively (if you use the kernel package in https://github.com/7Ji/archrepo)  so performance could be better.

The only thing needed, from my experience on Amlogic devices and my contribution to EmuELEC, was just some ARM proprietary driver blobs. If you need video decoding there's also already AUR packages built with mpp support.

If you could get it working I would be glad to maintain those additional packages in my package repo.

Integral-Tech, 2023-08-01:
> Sorry, I've never even connected my opi5 to any display. I just got opi5 for its performance to function as a native AArch64 build server for my various projects. And since it's under active heavy [...]

I find GPU is not detected in ```neofetch``` and ```archey4``` :(

7Ji, 2023-08-01:
Just wrote a PKGBUILD to repack that deb to meet the Arch file system hierarchy:
```
# Maintainer: 7Ji <pugokughin@gmail.com>

_drvname=libmali-valhall-g610-g6p0-x11
pkgname="${_drvname}-gbm"
pkgver=1.9
pkgrel=1
pkgdesc="Proprietary blob for ARM Valhall g610 g6p0, found on OrangePi 5"
url="https://github.com/orangepi-xunlong/rk-rootfs-build"
license=('proprietary')
_srcname="${_drvname}_${pkgver}-${pkgrel}_arm64.deb"
source=("${url}/raw/rk3588_packages_bookworm/libmali/${_srcname}")
noextract=("${_srcname}")
sha256sums=('717c8614d9c4e7593d05f42e702e6dca2edaa6af4c515cead7a2e51f4e4e7d6c')
arch=('aarch64')
makedepends=('dpkg')
options=(!strip)

prepare() {
    local _debroot="${srcdir}/debroot"
    mkdir -p "${_debroot}"
    dpkg-deb --extract "${srcdir}/${_srcname}" "${_debroot}"
}

package() {
    local copyright="/usr/share/doc/${pkgname}/copyright"
    local _debroot="${srcdir}/debroot"
    install -D -m 644 "${_debroot}${copyright}" "${pkgdir}${copyright}"
    
    install -d -m 755 "${pkgdir}/"{usr/lib,etc}
    cp -rv "${_debroot}/etc/"* "${pkgdir}/etc/"
    cp -rv "${_debroot}/usr/lib/aarch64-linux-gnu/"* "${_debroot}/lib/"* "${pkgdir}/usr/lib/"
    sed -i "s_/usr/lib/aarch64-linux-gnu/mali_/usr/lib/mali_" "${pkgdir}/etc/ld.so.conf.d/00-aarch64-mali.conf"
}
```

[libmali-valhall-g610-g6p0-x11-gbm-1.9-1-aarch64.pkg.tar.gz](https://github.com/7Ji/orangepi5-archlinuxarm/files/12233240/libmali-valhall-g610-g6p0-x11-gbm-1.9-1-aarch64.pkg.tar.gz)

DaveWK, 2023-08-14:
Following up on this because I just was tinkering with this as well --

From a plain install it seemed to partially detect the mali GPU, but didn't have any OpenCL/Vulkan support. I `wget`-ed all the files from JeffyCN's libmali mirror: https://github.com/JeffyCN/mirrors/tree/libmali/lib/aarch64-linux-gnu for `g610-g13p0` into `/usr/lib`, and still didn't have vulkan support. I then installed https://aur.archlinux.org/packages/mesa-panfork-git for panfork, and now at least am seeing KDE Plasma report that it is not using `llvmpipe` for rendering now, and instead is using `Mali G610` 

From my (limited) understanding, panfork is "fake hardware vulkan" and uses some hackery with `swrast` to use hardware in a limited capacity for vulkan (I assume this is why the AUR doesn't build `panfrost` in the vulkan meson option?). As I understand it the Mali G610 does not have full hardware vulkan support at this time. 

However, there is work underway to add a new driver: https://www.collabora.com/news-and-blog/news-and-events/pancsf-a-new-drm-driver-for-mali-csf-based-gpus.html
which I believe has changed named to "panthor" and is hosted here: https://gitlab.freedesktop.org/panfrost/mesa/-/tree/v10+panthor

Not sure if this is usable yet, as I haven't been able to build it and have it produce working vulkan for the G610, but it may be due to my limited understanding of which build options to select for meson. Overall, the overall the status of vulkan on Mali G610 seems to be undocumented, other than a brief note on the ARM firmware page claiming that the G610 "is great for vulkan" despite lacking any indication of how to enable it for wayland.

7Ji, 2023-08-17:
So after I got a new opi5+, my opi5 is finally librated from its several months of heavy workloads and I tinkered with some GPU stuffs. The acutal parts needed are less than what I thought.

I've added a part on how to enable 3d acceleration to readme, and a part on how to do hardware encoding/decoding.

https://github.com/7Ji/orangepi5-archlinuxarm#gpu--3d-acceleration

7Ji, 2023-08-17:
If you want to use the proprietary blob drivers, I also packed them: https://aur.archlinux.org/pkgbase/libmali-valhall-g610 , they're also available from my repo:
```
sudo pacman -Syu libmali-valhall-g610-{dummy,gbm,wayland-gbm,x11-gbm,x11-wayland-gbm}
```
(Multiple variants of the libs can co-exist)

To use the blob drivers, you'll need to pre-load them via `LD_LIBRARY_PATH` (I didn't made them system-wide libraries, as they don't provide OpenGL but only OpenGLES, they'll break your DE running on OpenGL) like the following:
```
LD_LIBRARY_PATH=/usr/lib/mali-valhall-g610/x11-gbm [program]
```

Addtionally, if your program runs only on GL, not GLES (like Minecraft), `gl4es` must also be pre-loaded to translate GL calls to GLES calls:
```
LD_LIBRARY_PATH=/usr/lib/gl4es:/usr/lib/mali-valhall-g610/x11-gbm [program]
```

But the blobs really are a hit-or-miss, a quick performance comparison on vanilla Minecraft 1.16.5:
  - panfork mesa's panfrost: ~10 fps
  - panfork mesa's panfrost + `PAN_MESA_DEBUG=gofaster` environment: ~35 fps
  - g13p0 blob driver, `-x11-gbm` variant, + gl4es : ~21 fps

Like how I packed the blobs, you should stick with `panfork` as your system OpenGL library, and only use the blobs when you're running some GLES programs that can take advantage of them.

7Ji, 2023-08-21:
OK, I got the final piece in the Jigsaw puzzle: chormium with hwdec. 

A demo with 8K AV1  video:
[image]

I'll upload the patched chromium to AUR and pre-built image to my repo after I fix another minor issue (cucrently chromium will only find v4l lib under `/usr/lib64`, so a link `/usr/lib64` -> `/usr/lib` is needed, which is against the Arch file system hierarchy). For now, you can test the package by building https://github.com/7Ji-PKGBUILDs/chromium-mpp by yourself.

7Ji, 2023-08-21:
Closed by 2a6f640f564920433176cccac088da5812715b0f

The patched chromium package is available on AUR: https://aur.archlinux.org/packages/chromium-mpp , and is also available as pre-built package form my repo: https://github.com/7Ji/archrepo .

With this, we can finally say everything GPU-related is doable on not only OPi but also all RK3588 platforms, and it does that very nicely.