7Ji/orangepi5-archlinuxarm #6: OPI-5B: Booting from onboard eMMC, Flashing with rkdeveloptool

DaveWK, 2023-08-14:
Hi,
I have been working on trying to boot from eMMC and use that as my primary storage device. 
My initial attempt was not successful, however when I got a working image on an SD Card and updated using your 7ji AUR, I did notice the eMMC was now visible as a block device and I could see my failed attempt did successfully flash to the eMMC. Not sure if there is a driver that I need to include with uBoot or I messed up a step.

Steps I took:
Started with MASKROM button, successfully ran `rkdeveloptool db rk3588_spl_loader_v1.12.112.bin`
`rkdeveloptool ef`
`rkdeveloptool wl 0x0 ArchLinuxARM-aarch64-OrangePi5-20230731_142718.img`
`rkdeveloptool wl 0x40 idbloader.img`
`rkdeveloptool wl 0x4000 u-boot.itb`

but the device failed to boot.

I am not sure if the subsequent `rkdeveloptool wl` is breaking the partition table on the arch image, or if it was a driver issue for loading the eMMC in uboot. I was not sure if the image has allocated blank spaces for the bootloader or if I should be flashing this at a different offset to give space.

I was thinking of using `dd` to write the bootloader to `ArchLinuxARM-aarch64-OrangePi5-20230731_142718.img` so I could flash it all in 1 `rkdeveloptool wl 0x0` operation to avoid corrupting partitions, but wanted to check.
Would I instead run:
`dd if=idbloader.img of=ArchLinuxARM-aarch64-OrangePi5-20230731_142718.img seek=64`
`dd if=u-boot.itb of=ArchLinuxARM-aarch64-OrangePi5-20230731_142718.img seek=16384`

I got these u-boot files from the orangepi OS image. Alternatively I looked into using the armbian `deb` but I do not understand the `rksd_loader.img` file, and there is no `idbloader.img` file, so I wasn't sure what offset I should be flashing `rksd_loader.img` to, or if I needed a `trust.img` file

7Ji, 2023-08-14:
My image does not have bootloaders, and writing it to eMMC directly would mean breaking your existing bootloader on eMMC. You would need to use `-root.tar` instead and extract it to eMMC partitions created by yourself.

Note always, the main realease in this project is those kernel packages, and the rootfs archive, the `*.img` is just a shortcut if you don't want to go the Arch way to either `pacstrap` with kernel pacakges, or `bsdtar` extracting the rootfs archive like official `Arch Linux ARM`, in both case you should take care of the bootloader by yourself. `*.img` therefore does not contain the bootloader and expects your existing bootloader.

Addtionally, at least for my newly bought opi5+, there're some quirks:
 - If there's bootloader on SPI flash, the eMMC would be completely disabled and neither seen by u-boot nor kernel. I think this is due to eMMC reusing some of the SPI lines and the mux directing those lines to SPI would mean eMMC not being recognized.
 - The bootloader would need to be stored on the eMMC's user partition (e.g. `mmcblk1`) instead of the boot partitions (e.g. `mmcblk1boot0` and `mmcblk1boot1`), it therefore makes the first 4MiB not writable and you must skip those parts when you create your partitions:
    ```
    ~> sudo fdisk -l /dev/mmcblk1
    Disk /dev/mmcblk1: 232.96 GiB, 250139901952 bytes, 488554496 sectors
    Units: sectors of 1 * 512 = 512 bytes
    Sector size (logical/physical): 512 bytes / 512 bytes
    I/O size (minimum/optimal): 512 bytes / 512 bytes
    Disklabel type: gpt
    Disk identifier: BB3FDB43-B5B7-4246-A919-BE81F982EE19
    
    Device          Start       End   Sectors   Size Type
    /dev/mmcblk1p1     64      1023       960   480K Linux filesystem
    /dev/mmcblk1p2   1024      7167      6144     3M Linux filesystem
    /dev/mmcblk1p3   8192    212991    204800   100M EFI System
    /dev/mmcblk1p4 212992 488552447 488339456 232.9G Linux root (ARM-64)
    ```
    You'll need to set the partition you would want to use for `/boot` as `EFI System` or mark it bootable. As u-boot would search either first partition or such marked partition for boot files.

DaveWK, 2023-08-14:
Hmm, I am wondering about the SPI/EMMC distinction here -- I thought the 5B only had eMMC, and the 5-Original had SPI, but no eMMC (at least that's what orange pi's website has me to believe); when I use `rdeveloptool` it does report the size as the full 256 gig drive, so I assume in MASKROM it's indeed getting direct write access.

So I think my issue is that `ArchLinuxARM-aarch64-OrangePi5-20230731_142718.img` might put the partition table right at 0x0, then the boot partition on the subsequent sectors without leaving a gap -- I try to avoid whatever method makes you end up with `mmcblk1boot0/mmcblk1boot1` devices because I find it overly complicated and unhelpful compared to just leaving a ~32 MB gap of space (not a big deal for a 256 GB emmc) for the uboot/itbloader in the image I am flashing, and can update u-boot by just dd-ing to the 64/16384 offsets on the device without anything attempting to abstract the true raw offsets in the OS.

So I think what I'll need to do is `dd` the boot/root partitions from the `.img`, create a new `img` file with a partition table that leaves a 32 MB gap, then append the arch boot/root partitions after the gap.

7Ji, 2023-08-14:
You only need a 4MiB gap, as I showed in the above log.

You always end up with `mmcblk1boot0` and `mmcblk1boot1`, they're always there. They're partitions but not partitions users can decide. The manufacturers create those partitions at factory. So it's a pity OPi decide not to use them when they're already there:  wasting space on user partition is not a big deal but it makes it hard to partition the user space as you must take care not to overwrite the areas where the bootloader is stored.

As showed in the log, the area where the bootloader is stored is the first 7167 sectors, rounding it up to 8192 and make sure all your partitions (starting from p3) starts from sector offset 8192, i.e. 4MiB, is enough.

7Ji, 2023-08-14:
> So I think what I'll need to do is dd the boot/root partitions from the .img, create a new img file with a partition table that leaves a 32 MB gap, then append the arch boot/root partitions after the gap.

You don't need to do that, the opi5/5b/5+'s u-boot supports USB mass storage gadget mode, you can just do it the following way:
 - Use `rkdevtool` to only burn the bootloader onto eMMC
 - Boot the board and enter u-boot
 - Use `ums` command in u-boot to emulate the device as a USB Mass Storage device to your PC
 - Partition the USB Mass Storage, and write stuffs into it in any way you like.

DaveWK, 2023-08-14:
Ah neat -- I didn't know about `ums` mode -- that's much less time consuming than copying around loopback devices like I had been; thanks!

closed 2023-08-14 by DaveWK