7Ji/orangepi5-archlinuxarm #8: i2c loop at first startup

niels1voo, 2023-09-12:
I burned [ArchLinuxARM-aarch64-OrangePi5-20230912_021419-rkloader-5_sata.img.gz](https://github.com/7Ji/orangepi5-archlinuxarm/releases/download/nightly/ArchLinuxARM-aarch64-OrangePi5-20230912_021419-rkloader-5_sata.img.gz) to a samsung 2280 nvme drive. Inserted into my Orange Pi 5 Pluse and on boot I get stuck at:

rk3x-i2c fec80000.i2c: timeout, ipd:0x80, state: 3

Any idea how to work this out?

7Ji, 2023-09-13:
Use the 5 plus image, that image is for 5 non plus

niels1voo, 2023-09-13:
Thanks for the tip. Though not a total noob, it's been a while since I tinkered with Linux machines. Appreciate the thorough explanation.

7Ji, 2023-09-13:
I'll need to download the image to test on my 5 plus locally, the images I built locally (what I'm using) do not have such issue. I'm doubting there's something wrong during the FDT substitution, as that's a familar error when using rk3588s DTB on rk3588 device

7Ji, 2023-09-13:
latest nightly, `ArchLinuxARM-aarch64-OrangePi5-20230913_021608-rkloader-5_plus.img.gz`, burnt on SD card, on 5 plus, boot successful. 

And those messages are just normal kernel logs. The SBC is not stuck.

```
[  OK  ] Reached target Graphical Interface.

Arch Linux 5.10.110-6-aarch64-orangepi5+ (ttyFIQ0)

alarm login: alarm
[   21.023498] ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
Password: 
[alarm@alarm ~]$ uptime
 14:00:44 up 0 min,  1 user,  load average: 0.27, 0.06, 0.02
[alarm@alarm ~]$ [   42.241899] vcc_mipicsi0: disabling
[   42.242014] vcc_mipicsi1: disabling
[   42.242056] vcc_mipicsi1: disabling
echo hello?
hello?
[alarm@alarm ~]$
```
I guess you're having a bad wire between your host Tx and the onboard Rx?

closed 2023-09-13 by 7Ji

niels1voo, 2023-09-14:
Just noob stuff, nothing wrong with it after all.