7Ji/orangepi5-archlinuxarm #9: i am noob (((

InfernalWolf666, 2023-09-30:
Hello my dear friend! I'm a complete noob in the world of Linux (((( .
I have an orange pi 5 plus + emmc 256 GB + original wifi bluetooth from orange pi r 6.

I'm reading your manual, it's a little difficult for me.

I have a Windows computer. I use software from the manufacturer.

Could you make some instructions for complete noobs?

InfernalWolf666, 2023-09-30:
I'm reporting!
I flashed spi memory - I used your bootloader!
I installed the Linux image on ems

Lo and behold, numbers and letters started running! Wow !

But I don’t know what to do next, do I need to somehow add a working environment?

InfernalWolf666, 2023-09-30:
What is the password for the user "ALARM"?

InfernalWolf666, 2023-09-30:
Login : alarm
Password : alarm_please_change_me

7Ji, 2023-10-01:
What's already covered on Arch Wiki won't be documented here. The README of this project only documents project-specific instructions. Users are expected to have experience using Arch itself, or at least generic Linux distros. 

It seems complicated because it is complicated, and the details should not be hidden from you. If a friendly all-in-one helper script leads you through all the deployment process, then one day the parts you skipped will come back to bite you.

closed 2023-10-08 by 7Ji