7Ji/orangepi5-archlinuxarm #11: OPI 5 PLUS - KDE Plasma Waylland System Freezed (Black Screen) After Entering Password

JFLim1, 2023-11-07:
Flashed https://github.com/7Ji/orangepi5-archlinuxarm/releases/download/nightly/ArchLinuxARM-aarch64-OrangePi5-20231107_021931-rkloader-5_plus.img.gz to a SD Card.

Boot up successfully. 
Proceed to install KDE Plasma using the following command:

`sudo pacman -S xorg plasma plasma-wayland-session dolphin konsole kate kwrite kcalc`

Reboot.  At SDDM login screen select "Wayland", Enter Password and it will Freeze with a Black Screen.  Keyboard and Mouse not responding.

Login with Plasma-X11

Install GPU/3D Acceleration

`sudo pacman -Syu mesa-panfork-git mali-valhall-g610-firmware`

Reboot.  Select Plasma-Wayland and Enter Password.  Freeze with A Black Screen.  Keyboard and Mouse not responding.

Reinstall "plasma-wayland-session".  Same result cannot login to Plasma-Wayland.

Can login to Plasma-X11.

Is there any other packages needed to be installed?

JFLim1, 2023-11-08:
> Plasma + Wayland could not work, which is expected. My personal setup before using both my OPi5 and OPi5 Plus as servers was Plasma + X11.

Read that some users managed to give KDE Plasma Wayland to work on Ubuntu-22.04.3:  
https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/248
```
Hello, I tried various things and finally the login became faster.
However, I don't know what worked for you, but you can try it if you like.

sudo systemctl status sddm
Please check if it is started with plasma wayland.
If it is X11, please select plasma wayland when logging in (Not now).
/etc/profile.d/qt.sh edit
export QT_QPA_PLATFORM=wayland
reinstall sddm and plasma-workspace-wayland
reboot
4.login and change wallpaper
reboot
```

The kernel used is different though 5.10.160-17.

On Amlogiic S922X (GT King Pro) my experience is that it runs much better on Manjaro-Arm-KDE Wayland than on KDE-X11.  Wonder whether it is going to be the same for OPI 5 Plus.

Would like to try KDE Plasma-Wayland on OPI 5 Plus.  Any tips or advise on how to achieve this is much appreciated.  Thank you.

7Ji, 2023-11-08:
> 5.10.160-17

is probably not related, as this is hardly a kernel issue. 

Also the kernel package is strictly kept in line with OrangePi's source [here](https://github.com/orangepi-xunlong/linux-orangepi/tree/orange-pi-5.10-rk3588). 5.10.110 is also the latest Rockchip BSP for RK3588, whereas 5.10.160 is [Radxa's  own kernel tree](https://github.com/radxa/kernel/tree/linux-5.10-gen-rkr4.1) with added RK3588 support (It's a hack on top of Android kernel already, and then running it on OPi5 is another hack). Comparing vendor kernel's version just makes no sense.

For now I don't really care about graphical experience on this SoC any more as I don't have that much time. I would rather wait for mainline v6.7 than to put more effort on vendor kernels.

wyf9661, 2023-11-19:
why mentiond kernel v6.7, did you mean that in v6.7 we can use maintain mesa driver for g610 instead of panfork version?

7Ji, 2023-11-19:
v6.7 should be stable and patch-less to run on RK3588. Currently there are some mainline-ish kernel packages scattered around the community and many claimed they run, but those are really just band-aid kernels with tens of patches cherry-picked. I don't like old kernel + new patch + self-maintained way, that combination is really a welcome to lagged update and security concerns (to be honest, why not just use Linus' master branch in that case?) which is why I insist on using vendor kernel in this project currently.

Panthor is the upcoming kernel driver that'll support CSF-based GPUs. This is a whole new open-source ARM GPU driver landing in mainline kernel after many years, after lima and panfrost. It was developed by Mesa maintainer and with it added, you could expect finally stable and smooth graphics experience that does not have trivia here and there, all powered by Mesa. 

Many exciting changes about that driver is happening on LKML: https://lore.kernel.org/all/?q=panthor

The most recent tag Linus tagged was v6.7-rc1, 6 days ago, and usually it takes 7 release candidates to reach a final release, so there's still 6 weeks to go, and most of those panthor patches should be merged before that, so we can expect a stable panthor experience.

wyf9661, 2023-11-29:
I try to bump my kernel to v6.7, all things goes well except hdmi output, but I still think it's a great new beginning. Now I'm looking forward for the new version.  thanks a lot for your patient explanation!
[image]