7Ji/orangepi5-archlinuxarm #12: pacman -Syu ffmpeg-mpp --error: failed retrieving file 'ffmpeg-mpp-2.6.0-9-aarch64.pkg.tar.zst' from github.com

JFLim1, 2023-11-07:
Unable to install "ffmpeg-mpp" as per suggestion: https://github.com/7Ji/orangepi5-archlinuxarm

```
[alarm@alarm ~]$ sudo pacman -Syu ffmpeg-mpp
[sudo] password for alarm: 
:: Synchronizing package databases...
 core                  233.7 KiB  77.1 KiB/s 00:03 [########################] 100%
 extra                   9.1 MiB  4.74 MiB/s 00:02 [########################] 100%
 alarm is up to date
 aur is up to date
 7Ji is up to date
:: Starting full system upgrade...
resolving dependencies...
looking for conflicting packages...
:: ffmpeg-mpp and ffmpeg are in conflict. Remove ffmpeg? [y/N] y

Packages (11) aom-3.7.0-1  cjson-1.7.16-1  ffmpeg-2:6.0-13 [removal]
              gawk-5.3.0-1  librist-0.2.10-1  libyuv-r2426+464c51a0-1
              mbedtls-3.4.1-1  onevpl-2023.3.1-1  rav1e-0.6.6-3  sudo-1.9.15-1
              ffmpeg-mpp-2:6.0-9

Total Download Size:   18.90 MiB
Total Installed Size:  72.09 MiB
Net Upgrade Size:      28.26 MiB

:: Proceed with installation? [Y/n] Y
:: Retrieving packages...
 ffmpeg-mpp-2.6.0-9-aarch64.pkg.tar.zst failed to download
 Total ( 1/10)          18.9 MiB  55.6 MiB/s 00:00 [########################] 100%
error: failed retrieving file 'ffmpeg-mpp-2.6.0-9-aarch64.pkg.tar.zst' from github.com : The requested URL returned error: 404
warning: failed to retrieve some files
error: failed to commit transaction (failed to retrieve some files)
Errors occurred, no packages were upgraded.
[alarm@alarm ~]$
```

7Ji, 2023-11-07:
The package was broken for almost half a month since `texinfo` updated. The repo builder is set to only keep the latest package which links to the latest dependencies, and since it can't build `ffmpeg-mpp` with latest `texinfo` as dependency, the package was removed.

7Ji, 2023-11-07:
I've manually locked the last successful build, and now `pacman -Syu ffmpeg-mpp` should just work

closed 2023-11-07 by 7Ji

JFLim1, 2023-11-08:
> I've manually locked the last successful build, and now `pacman -Syu ffmpeg-mpp` should just work

Thank you.  

`sudo pacman -S ffmpeg-mpp
`
works now.

Will try to get Chromium-mpp to be able to stream Youtube with vpu hw acceleration by following your guide.

Thank you.

JFLim1, 2023-11-08:
Hi @7Ji,

Question: Is mali blobs needed to install first before Chromium-mpp can stream with vpu hw acceleration?  Wasn't clear on this.

`LD_LIBRARY_PATH=/usr/lib/mali-valhall-g610/x11-gbm chromium --use-gl=egl`

Currently did not install Mali Blobs had installed:

`sudo pacman -Syu mesa-panfork-git mali-valhall-g610-firmware`

As such thre is NO 
`/usr/lib/mali-valhall-g610/x11-gbm`

7Ji, 2023-11-08:
Chromium-mpp works without the blob, but without hwdec

To get hwdec, you must also install the Mali blob driver, and run chormium-mpp with the x11 blob driver.

JFLim1, 2023-11-08:
> To get hwdec, you must also install the Mali blob driver, and run chormium-mpp with the x11 blob driver.

Thank you. Noted.

Hi @7Ji,

Installed mpv using 

`sudo pacman -S mpv`

But mpv does not seems to support "--hwdec=auto".

```
[alarm@alarm ~]$ mpv --hwdec=auto https://www.youtube.com/watch?v=aqz-KE-bpKQ
 (+) Video --vid=1 (*) (vp9 3840x2160 60.000fps)
 (+) Audio --aid=1 (*) (aac 6ch 48000Hz)
File tags:
 Uploader: Blender
 Channel_URL: https://www.youtube.com/channel/UCSMOQeBJ2RAnuFungnQOxLg
Failed to open VDPAU backend libvdpau_rockchip.so: cannot open shared object file: No such file or directory
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize MPP context (code = -1).
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize RKMPP Codec.
Could not open codec.
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize MPP context (code = -1).
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize RKMPP Codec.
Could not open codec.
Decoder init failed for vp9_rkmpp_decoder
Failed to open VDPAU backend libvdpau_rockchip.so: cannot open shared object file: No such file or directory
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize MPP context (code = -1).
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize RKMPP Codec.
Could not open codec.
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize MPP context (code = -1).
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize RKMPP Codec.
Could not open codec.
Decoder init failed for vp9
Failed to open VDPAU backend libvdpau_rockchip.so: cannot open shared object file: No such file or directory
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize MPP context (code = -1).
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize RKMPP Codec.
Could not open codec.
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize MPP context (code = -1).
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize RKMPP Codec.
Could not open codec.
Decoder init failed for vp9_v4l2m2m
Failed to open VDPAU backend libvdpau_rockchip.so: cannot open shared object file: No such file or directory
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize MPP context (code = -1).
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize RKMPP Codec.
Could not open codec.
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize MPP context (code = -1).
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize RKMPP Codec.
Could not open codec.
Decoder init failed for libvpx-vp9
Failed to open VDPAU backend libvdpau_rockchip.so: cannot open shared object file: No such file or directory
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize MPP context (code = -1).
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize RKMPP Codec.
Could not open codec.
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize MPP context (code = -1).
[ffmpeg/video] vp9_rkmpp_decoder: Failed to initialize RKMPP Codec.
Could not open codec.
Decoder init failed for vp9_qsv
Failed to initialize a decoder for codec 'vp9'.
Video: no video

Exiting... (Errors when loading file)
[alarm@alarm ~]$ 
```

7Ji, 2023-11-08:
Basically for anything that renders to the actual screen you needs the blob driver and run the executable with the blob driver. 

Arch does not split packages like Debian-like do. Ofc there's no libavcodec, it is a part of ffmpeg. You install ffmpeg, you install ffmpeg as a whole.

JFLim1, 2023-11-08:
> Basically for anything that renders to the actual screen you needs the blob driver and run the executable with the blob driver.

How to get mpv to have vpu hw acceleration?  Installing "ffmpeg-mpp" not sufficient to have vpu hw acceleration on mpv?

7Ji, 2023-11-08:
> you need the blob driver and run the executable with the blob driver

Like how you do it for chromium, prefix your command with LD_...

There's also a wrapper `libmali` that'll do it for you if you don't want to type those each time, just prefix the command with `libmali`

Also I don't use MPV nor do I use OPi5/Plus for desktop. When I used them for desktop I used VLC.

JFLim1, 2023-11-08:
Hi @7Ji,

Read from https://forum.radxa.com/t/archlinux-on-rock5b/13851 to get mpv to use "mpp" need to perform the following:

> Video Decoder Accelaration with FFMpeg & RKMPP
> Install the packages mpp-git, ffmpeg4.4-mpp, ffmpeg-mpp, kodi-stable-mpp-git.
> 
> Make sure that your user is added to video group. This is neceesary for user space applications to access mpp drivers other-wise, video acceleration will not work. You can verify this by issuing groups and check if the video group is there, if not, you can add your user with usermod -A -g video myuser. Please note that after a user is added to a new group, restart is required to make it active.

But somehow "usermod -A -g video alarm" is not a valid command.

```
[alarm@alarm ~]$ sudo usermod -A -g video alarm
[sudo] password for alarm: 
usermod: invalid option -- 'A'
Usage: usermod [options] LOGIN

Options:
  -a, --append                  append the user to the supplemental GROUPS
                                mentioned by the -G option without removing
                                the user from other groups
  -b, --badname                 allow bad names
  -c, --comment COMMENT         new value of the GECOS field
  -d, --home HOME_DIR           new home directory for the user account
  -e, --expiredate EXPIRE_DATE  set account expiration date to EXPIRE_DATE
  -f, --inactive INACTIVE       set password inactive after expiration
                                to INACTIVE
  -g, --gid GROUP               force use GROUP as new primary group
  -G, --groups GROUPS           new list of supplementary GROUPS
  -h, --help                    display this help message and exit
  -l, --login NEW_LOGIN         new value of the login name
  -L, --lock                    lock the user account
  -m, --move-home               move contents of the home directory to the
                                new location (use only with -d)
  -o, --non-unique              allow using duplicate (non-unique) UID
  -p, --password PASSWORD       use encrypted password for the new password
  -P, --prefix PREFIX_DIR       prefix directory where are located the /etc/* files
  -r, --remove                  remove the user from only the supplemental GROUPS
                                mentioned by the -G option without removing
                                the user from other groups
  -R, --root CHROOT_DIR         directory to chroot into
  -s, --shell SHELL             new login shell for the user account
  -u, --uid UID                 new UID for the user account
  -U, --unlock                  unlock the user account
  -v, --add-subuids FIRST-LAST  add range of subordinate uids
  -V, --del-subuids FIRST-LAST  remove range of subordinate uids
  -w, --add-subgids FIRST-LAST  add range of subordinate gids
  -W, --del-subgids FIRST-LAST  remove range of subordinate gids
```

This is embarrassing, how to add user "alarm" to video group?

7Ji, 2023-11-08:
It's -aG not -Ag, and you need to relogin after adding that for the supplemental groups to take effect

JFLim1, 2023-11-08:
> It's -aG not -Ag, and you need to relogin after adding that for the supplemental groups to take effect

Thanks a lot.  Will see with this whether mpv will be able to access "mpp drivers".

JFLim1, 2023-11-08:
> It's -aG not -Ag, and you need to relogin after adding that for the supplemental groups to take effect

Ok, getting some improvement but still not there yet. Complaining **"(0 layers and 0 planes, but expected 2 planes)"**

> [alarm@alarm ~]$ sudo usermod -a -G video $LOGNAME
> [alarm@alarm ~]$ groups
> video wheel
> [alarm@alarm ~]$

```
mpv --hwdec=auto https://www.youtube.com/watch?v=aqz-KE-bpKQ
[vo/gpu/drmprime] Mapped surface with format 'nv12' has unexpected number of planes. (0 layers and 0 planes, but expected 2 planes)
[vo/gpu/drmprime] mapping DRM dmabuf failed
[vo/gpu] Mapping hardware decoded surface failed.
```
Video play to a "Blank Blue Screen", Sound is available and "HTOP" shows low CPU usage meaning with vpu hw acceleration but a Blank Blue Screen.

Looks like config issue or missing a package?