7Ji/orangepi5-archlinuxarm #13: "error: not enough free disk space"  when updating kernel

Dmitrij-Solovjev, 2023-11-15:
I have a problem updating the Linux kernel. I trying install kodi (kodi-rpi), for example, but it requires install enother linux kernel (linux-aarch64-6.2.10-1), for example. But the installation is interrupted due to an error:
```
error: Partition /boot too full: 204079 blocks needed, 88247 blocks free
error: not enough free disk space
error: failed to commit transaction (not enough free disk space)
Errors occurred, no packages were upgraded.
```
my disk part:

```
mmcblk0      179:0    0    58G  0 disk 
|-mmcblk0p1  179:1    0   480K  0 part 
|-mmcblk0p2  179:2    0     3M  0 part 
|-mmcblk0p3  179:3    0   3.4G  0 part /boot
`-mmcblk0p4  179:4    0  54.5G  0 part /
```

I was able to bypass the existing limitation by installing the package kodi-rpi-git with linux-rpi, but this still did not allow the installation to complete, since the /boot partition was completely full and the kernel was not updated

7Ji, 2023-11-15:
Install `kodi-matrix-mpp-git` and `kodi-nexus-mpp-git`, they're from my repo

The `kodi(-rpi)` you're installed is only for rpi

closed 2023-11-15 by 7Ji

wyf9661, 2023-11-16:
https://www.zhihu.com/question/62428599 @Dmitrij-Solovjev