7Ji/orangepi5-archlinuxarm #15: kde wayland

wyf9661, 2023-11-16:
Hi, I really appreciate you for providing this images for orangepi, when I use panfork mesa driver in kde(X11), sometimes the desktop crashed and I  restarted sddm then it will be ok.
Some people told me that the panfork driver was made to support wayland, so it may not perform well on x11.But when I change X11 to wayland on KDE, the plasma does not work, could you please help me ? 
Best wishes, thank you!

7Ji, 2023-11-16:
Panfork + plasma + wayland = nogo, it simply won't work, and there's no way to work around it. Using gnome on wayland is OK.

I'm using both my OPi5 plus and OPi5 as build servers for https://github.com/7Ji/archrepo , and they work like 7x24 to keep all packages up-to-date, so I can't dig again on actual hardware.

When I did use OPi5 for desktop, my setup was plasma + x11. And when it hangs I just call dbus to stop plasma or killall if that failed.

closed 2023-11-16 by wyf9661

7Ji, 2023-11-20:
FYI for a quick restart of plasma shell if it freezes:

```
killall plasmashell
kstart5 plasmashell
```

wyf9661, 2023-11-21:
thank you.when it freezes,this method really does well.but i am wondering is there any way on the situation when the sreen is locked and does not reacted any more sometimes.when i try to restart sddm, it faied. the only thing i can do is to reboot.

wyf9661, 2023-11-22:
You are right. I mean when i use win + L to lock screen several times, the screen did not react even though when i move my mouse, it moves on screen, but the graph does not change. 
And then, I use ctrl + alt + F1/2/3/... to switch tty terminal, input systemctl restart sddm, it failed. And now, i usually use reboot -f to restart the PC, but i don't know if there be a better way.
Best Wishes.
[image]

[image]

wyf9661, 2023-11-22:
[image]

wyf9661, 2023-11-22:
[image]

wyf9661, 2023-12-14:
By coincidence, I know why there is nobody meets things like me, For me, I use two monitor for work. When I turn to using one screen and shut down another, this situation does not hit me anymore.

7Ji, 2023-12-14:
> When I turn to using one screen and shut down another, this situation does not hit me anymore.

The 5/5B has a HDMI + a DP over usb-c, and the 5Plus has an additional HDMI. I wonder whether it's the dual-screen setup itself causing problem, or displaying over either of the HDMI/DP port is troublesome. Is single HDMI OK? And/Or is single DP OK?

wyf9661, 2023-12-14:
I usyally use to HDMI for display, since I don't  have any typec2dp line. And I tested either of the HDMI port in single mode, both was ok.It seems when I locked the screen and click the mouse and make some move from one to another, it may cause no reaction for display.Maybe it is associated with the synchronization of two screen,I don't know.But the v6.7 will come and I don't care too much about the panfork mesa driver.
linux-rockchip has already bump to rc4, new days will come soon~

wyf9661, 2024-01-02:
[image]

wayland is supported in v6.1 now