7Ji/orangepi5-archlinuxarm #16: Hello! How should I transfer my installation to eMMC (Orange 5B)?

DragonSerW, 2023-11-16:
Hello 7Ji!
Please help me, how should I transfer my instance of orangepi5-archlinuxarm to eMMC?
I used to install ARch from image `ArchLinuxARM-aarch64-OrangePi5-20231116_022315-rkloader-5b.img`.

7Ji, 2023-11-16:
Just dd'ing that image onto eMMC is enough.

Remember to remove the current boot medium as they share the same filesystem identifier and mount could be randomly mixed.

DragonSerW, 2023-11-16:
Ok, I will try the dd command.
One more question: how should I resize the `/` partition at `mmcblk0` (microSD)?

7Ji, 2023-11-16:
You'll need another device, or on the same device but booting from different media. When no partition is mounted on the drive just run `fdisk` against that drive, remove part 4 and create a new part 4 with the same offset, allocate freely. After that `e2fsck -f [part]` and `resize2fs [part]`

7Ji, 2023-11-16:
Also, you probably want to partition the drive first and then extract from a `-root.tar` release, instead of dd'ing and resizing.

DragonSerW, 2023-11-16:
If I will remove `/dev/sda4` and re-create it using Gnome-disk-utility, should I perform `e2fsck -f /dev/sda4` and `resize2fs /dev/sda4`?

7Ji, 2023-11-16:
I didn't use any GUI partition tools on Linux before. To play it safely I'd recommend always do that. But, as partition != fs, even if the utility does not resize the filesystem alongside the partition, it wouldn't corrupt the fs and you could still mount the resized-or-not filesystem on it, and actually resize it later if it's not resized.

closed 2023-11-16 by 7Ji

7Ji, 2023-11-20:
That's a genral question and not related to my project. Just do it the Arch way.

7Ji, 2023-11-21:
OK I'll write a quick summary of what you should do without actual testing it, as the only board with eMMC in my hand is building packages for my repo 7x24. If there's anything going boom you better have a backup.

1. Run your system off SD
2. Go to https://github.com/7Ji/orangepi5-rkloader/releases/tag/nightly to download the corresponding rkloader
3. Write rkloader to eMMC `dd if=rkloader.img of=/dev/mmcblk1 bs=1M conv=sync`
4. Partition the eMMC, `fdisk /dev/mmcblk1`, there should already be two partitions, **DON'T EVER TOUCH THEM**. Create your boot partition as part 3 and root partition as part 4. The partition offset should be at least 8192 to avoid overwriting rkloader. Set your boot partition type as EFI System Partition.
5. Create FAT32 on boot and ext4 on root. `mkfs.vfat -F 32 /dev/mmcblk1p3`, `mkfs.ext4 -m 0 /dev/mmcblk1p4`
6. Get and write down the UUID of your boot and root partition, `blkid /dev/mmcblk1p{3,4}`
7. mount root to `/mnt` and boot to `/mnt/boot`
8. Clone your current installation onto eMMC  `rsync -avxHAXS --exclude={"/dev/*","/proc/*","/sys/*","/tmp/*","/run/*","/mnt/*","/media/*","/lost+found"} /. /mnt`
9. Update the UUID of root partition in `/mnt/boot/extlinux/extlinux.conf`, and root + boot in `/mnt/etc/fstab`
10. Chroot and recreate initramfs `arch-chroot /mnt mkinitcpio -P`
11. Umount and sync everything

DragonSerW, 2023-11-21:
> Run your system off SD
You mean, I have to start my system from microSD?
Or start it without microSD?

7Ji, 2023-11-22:
From

DragonSerW, 2023-11-29:
Hello 7Ji!
I made the 1-9 steps.
On step 10 there is an error:
```bash
[root@alarm ~]# arch-chroot /mnt mkinitcpio -P
==> Building image from preset: /etc/mkinitcpio.d/linux-aarch64-orangepi5.preset: 'default'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-aarch64-orangepi5 -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-aarch64-orangepi5.img
==> ERROR: specified kernel image does not exist: '/boot/vmlinuz-linux-aarch64-orangepi5'
==> Building image from preset: /etc/mkinitcpio.d/linux-aarch64-orangepi5.preset: 'fallback'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-aarch64-orangepi5 -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-aarch64-orangepi5-fallback.img -S autodetect
==> ERROR: specified kernel image does not exist: '/boot/vmlinuz-linux-aarch64-orangepi5'
[root@alarm ~]# 
```
And 1 more question: how Should I "unmount everything"?

7Ji, 2023-11-30:
> Hello 7Ji! I made the 1-9 steps. On step 10 there is an error:
> 
> ```shell
> [root@alarm ~]# arch-chroot /mnt mkinitcpio -P
> ==> Building image from preset: /etc/mkinitcpio.d/linux-aarch64-orangepi5.preset: 'default'
> ==> Using configuration file: '/etc/mkinitcpio.conf'
>   -> -k /boot/vmlinuz-linux-aarch64-orangepi5 -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-aarch64-orangepi5.img
> ==> ERROR: specified kernel image does not exist: '/boot/vmlinuz-linux-aarch64-orangepi5'
> ==> Building image from preset: /etc/mkinitcpio.d/linux-aarch64-orangepi5.preset: 'fallback'
> ==> Using configuration file: '/etc/mkinitcpio.conf'
>   -> -k /boot/vmlinuz-linux-aarch64-orangepi5 -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-aarch64-orangepi5-fallback.img -S autodetect
> ==> ERROR: specified kernel image does not exist: '/boot/vmlinuz-linux-aarch64-orangepi5'
> [root@alarm ~]# 
> ```
> 
> And 1 more question: how Should I "unmount everything"?

> mount root to /mnt and boot to /mnt/boot

That's not what you did

> And 1 more question: how Should I "unmount everything"?

Generic Linux question, `umount -R /mnt`, do not ask more.

DragonSerW, 2023-11-30:
Should I perform `sudo rsync -avxHAXS --exclude={"/dev/*","/proc/*","/sys/*","/tmp/*","/run/*","/mnt/*","/media/*","/lost+found"} /boot /mnt/boot`?

7Ji, 2023-11-30:
Why edit? The original command in mail was right