7Ji/orangepi5-archlinuxarm #17: Should I perform `pacstrap -K /mnt base linux linux-firmware-orangepi`?

DragonSerW, 2023-11-16:
```bash
pacstrap -K /mnt base linux linux-firmware-orangepi
```
or even
```bash
pacstrap -K /mnt base linux linux-firmware-orangepi5```
 - -should I perform it?
My installation media is `ArchLinuxARM-aarch64-OrangePi5-20231116_022315-rkloader-5b.img`.

7Ji, 2023-11-17:
Neither, most importantly not `linux`, that's for mainline (or claimed to be so, but it won't work on my mainline Amlogic devices)

run `pacman -Q` to get a list of installed packages

https://github.com/7Ji/orangepi5-archlinuxarm/blob/836f87929416ab1623788fe3d103f1c35de3adaf/cross_nobuild.sh#L212

https://github.com/7Ji/orangepi5-archlinuxarm/blob/836f87929416ab1623788fe3d103f1c35de3adaf/cross_nobuild.sh#L227

7Ji, 2023-11-17:
Closed by https://github.com/7Ji/orangepi5-archlinuxarm/commit/47441cb559f3845bce4d11738eb42c834c1c521c

https://github.com/7Ji/orangepi5-archlinuxarm#manual-installation-using-pacstrap

closed 2023-11-17 by 7Ji