7Ji/orangepi5-archlinuxarm #18: Telegram-desktop freezes all the system

DragonSerW, 2023-11-17:
`t-d ` requires `ffmpeg` lib, but I removed it and install `ffmpeg-mpp`.
Since that t-d and Firefox does not work properly.

7Ji, 2023-11-17:
Please provide detailed configuration, especially which display server/protocol + DE/WM + GL driver

Functionality-wise `ffmpeg-mpp` only provides more, not less. It should not break.

If you're sure it's the problem of that package (e.g. if replacing it back to `ffmpeg` would fix the problem), then open an issue on https://github.com/7Ji-PKGBUILDs/ffmpeg-mpp and ping me and that package's maintainer hbiyik

DragonSerW, 2023-11-17:
```bash
[DragonSerW@alarm ~]$ inxi -b
System:
  Host: alarm Kernel: 5.10.110-6-aarch64-orangepi5+ arch: aarch64 bits: 64
    Desktop: GNOME v: 45.1 Distro: Arch Linux ARM
Machine:
  Type: ARM System: Orange Pi 5 details: N/A
CPU:
  Info: 3x 4-core Model N/A [MCP AMP] speed (MHz): avg: 906
    min/max: 408/1800:2400
Graphics:
  Device-1: display-subsystem driver: rockchip_drm v: N/A
  Device-2: mali-bifrost driver: mali v: N/A
  Device-3: rk3588-dw-hdmi driver: dwhdmi_rockchip v: N/A
  Device-4: Logitech B525 HD Webcam driver: snd-usb-audio,uvcvideo type: USB
  Display: wayland server: X.org v: 1.21.1.9 with: Xwayland v: 23.2.2
    compositor: gnome-shell driver: N/A resolution: no compositor data
    resolution: 1920x1200
  API: OpenGL v: 4.5 compat-v: 3.0 vendor: panfrost v: N/A
    renderer: Mali-G610 (Panfrost)
Network:
  Device-1: Broadcom driver: pcieh
  Device-2: rk3588-gmac driver: rk_gmac_dwmac
  Device-3: wlan-platdata driver: wlan_platdata
Drives:
  Local Storage: total: 262.14 GiB used: 8.47 GiB (3.2%)
Info:
  Processes: 285 Uptime: 1h 3m Memory: total: N/A available: 15.6 GiB
  used: 1.89 GiB (12.1%) Shell: Bash inxi: 3.3.31
[DragonSerW@alarm ~]$ 
```

7Ji, 2023-11-20:
[image]
 They run just fine on a brand new installation

closed 2023-11-20 by 7Ji