7Ji/orangepi5-archlinuxarm #19: Hello! Please check the output of `top -b -n 1 > file.txt`

DragonSerW, 2023-11-18:
My question is: Chromium-mpp, even if it idle, utilise about 25 % CPU.
And, kgx utilise 19-25 % CPU.

```bash
top - 13:51:08 up 16:22,  1 user,  load average: 0,97, 1,59, 1,45
Tasks: 293 total,   1 running, 292 sleeping,   0 stopped,   0 zombie
%Cpu0  : 33,3 us,  6,7 sy,  0,0 ni, 53,3 id,  0,0 wa,  0,0 hi,  6,7 si,  0,0 st 
%Cpu1  : 30,8 us, 23,1 sy,  0,0 ni, 46,2 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st 
%Cpu2  : 15,4 us,  0,0 sy,  0,0 ni, 76,9 id,  0,0 wa,  0,0 hi,  7,7 si,  0,0 st 
%Cpu3  :  7,7 us, 15,4 sy,  0,0 ni, 76,9 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st 
%Cpu4  :  0,0 us,  0,0 sy,  0,0 ni,100,0 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st 
%Cpu5  :  0,0 us,  0,0 sy,  0,0 ni,100,0 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st 
%Cpu6  :  0,0 us,  0,0 sy,  0,0 ni,100,0 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st 
%Cpu7  :  0,0 us,  0,0 sy,  0,0 ni,100,0 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st 
MiB Mem :  15970,6 total,  11238,8 free,   2506,3 used,   2526,2 buff/cache     
MiB Swap:      0,0 total,      0,0 free,      0,0 used.  13464,3 avail Mem 

    PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
  17561 DragonS+  20   0   12488   5332   3252 R  23,1   0,0   0:00.06 top
   9357 DragonS+  20   0 1694316  34152  25740 S  15,4   0,2  10:24.57 pulseau+
  17488 DragonS+  20   0  170,7g 182040 109500 S  15,4   1,1   0:19.63 chromium
   8913 DragonS+  20   0 5661240 476540 387320 S   7,7   2,9  21:11.01 gnome-s+
  16627 DragonS+  20   0  648160  30604  23548 S   7,7   0,2   0:27.35 cmus
  16718 DragonS+  20   0 1623684 176568 121116 S   7,7   1,1   0:07.31 nautilus
      1 root      20   0   22076  12772   9584 S   0,0   0,1   0:13.45 systemd
      2 root      20   0       0      0      0 S   0,0   0,0   0:00.27 kthreadd
      3 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 rcu_gp
      4 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 rcu_par+
      8 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 mm_perc+
      9 root      20   0       0      0      0 S   0,0   0,0   0:00.00 rcu_tas+
     10 root      20   0       0      0      0 S   0,0   0,0   0:00.00 rcu_tas+
     11 root      20   0       0      0      0 S   0,0   0,0   0:16.57 ksoftir+
     12 root      20   0       0      0      0 I   0,0   0,0   0:46.24 rcu_sch+
     13 root      rt   0       0      0      0 S   0,0   0,0   0:00.06 migrati+
     14 root      20   0       0      0      0 S   0,0   0,0   0:00.00 cpuhp/0
     15 root      20   0       0      0      0 S   0,0   0,0   0:00.00 cpuhp/1
     16 root      rt   0       0      0      0 S   0,0   0,0   0:00.04 migrati+
     17 root      20   0       0      0      0 S   0,0   0,0   0:02.75 ksoftir+
     20 root      20   0       0      0      0 S   0,0   0,0   0:00.00 cpuhp/2
     21 root      rt   0       0      0      0 S   0,0   0,0   0:00.03 migrati+
     22 root      20   0       0      0      0 S   0,0   0,0   0:01.12 ksoftir+
     25 root      20   0       0      0      0 S   0,0   0,0   0:00.00 cpuhp/3
     26 root      rt   0       0      0      0 S   0,0   0,0   0:00.03 migrati+
     27 root      20   0       0      0      0 S   0,0   0,0   0:00.89 ksoftir+
     30 root      20   0       0      0      0 S   0,0   0,0   0:00.00 cpuhp/4
     31 root      rt   0       0      0      0 S   0,0   0,0   0:00.00 migrati+
     32 root      20   0       0      0      0 S   0,0   0,0   0:00.23 ksoftir+
     35 root      20   0       0      0      0 S   0,0   0,0   0:00.00 cpuhp/5
     36 root      rt   0       0      0      0 S   0,0   0,0   0:00.00 migrati+
     37 root      20   0       0      0      0 S   0,0   0,0   0:00.10 ksoftir+
     40 root      20   0       0      0      0 S   0,0   0,0   0:00.00 cpuhp/6
     41 root      rt   0       0      0      0 S   0,0   0,0   0:00.00 migrati+
     42 root      20   0       0      0      0 S   0,0   0,0   0:00.16 ksoftir+
     45 root      20   0       0      0      0 S   0,0   0,0   0:00.00 cpuhp/7
     46 root      rt   0       0      0      0 S   0,0   0,0   0:00.00 migrati+
     47 root      20   0       0      0      0 S   0,0   0,0   0:00.10 ksoftir+
     50 root      20   0       0      0      0 S   0,0   0,0   0:00.00 kdevtmp+
     51 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 netns
     60 root      20   0       0      0      0 S   0,0   0,0   0:00.00 kauditd
     61 root      20   0       0      0      0 S   0,0   0,0   0:00.00 oom_rea+
     62 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 writeba+
    106 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 kintegr+
    107 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 kblockd
    108 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 blkcg_p+
    109 root      20   0       0      0      0 S   0,0   0,0   0:00.28 kconsole
    110 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 tpm_dev+
    111 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 devfreq+
    112 root     -51   0       0      0      0 S   0,0   0,0   0:00.00 watchdo+
    115 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 rpciod
    117 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 xprtiod
[... 217 lines cut ...]
  16890 root       0 -20       0      0      0 I   0,0   0,0   0:00.11 kworker+
  16904 root       0 -20       0      0      0 I   0,0   0,0   0:02.35 kworker+
  16911 root      20   0       0      0      0 I   0,0   0,0   0:00.00 kworker+
  16912 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 kworker+
  17035 root       0 -20       0      0      0 I   0,0   0,0   0:00.17 kworker+
  17037 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 kworker+
  17038 root       0 -20       0      0      0 I   0,0   0,0   0:00.02 kworker+
  17040 root      20   0       0      0      0 I   0,0   0,0   0:00.02 kworker+
  17041 root      20   0       0      0      0 I   0,0   0,0   0:00.01 kworker+
  17042 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 kworker+
  17060 root       0 -20       0      0      0 I   0,0   0,0   0:00.04 kworker+
  17392 root      20   0       0      0      0 I   0,0   0,0   0:00.59 kworker+
  17394 root       0 -20       0      0      0 I   0,0   0,0   0:00.52 kworker+
  17423 DragonS+  20   0 1524120 165296 107000 S   0,0   1,0   0:14.67 gnome-t+
  17475 root       0 -20       0      0      0 I   0,0   0,0   0:00.02 kworker+
  17507 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 kworker+
  17508 DragonS+  20   0  170,7g  66332  47796 S   0,0   0,4   0:00.05 chromium
  17525 root       0 -20       0      0      0 I   0,0   0,0   0:00.00 kworker+
  17526 root       0 -20       0      0      0 I   0,0   0,0   0:00.58 kworker+
  17547 root       0 -20       0      0      0 I   0,0   0,0   0:00.07 kworker+
  17548 root       0 -20       0      0      0 I   0,0   0,0   0:00.02 kworker+
  17551 root      20   0       0      0      0 I   0,0   0,0   0:00.00 kworker+
  17552 root      20   0       0      0      0 I   0,0   0,0   0:00.00 kworker+
  17559 root      20   0       0      0      0 I   0,0   0,0   0:00.00 kworker+
```

7Ji, 2023-11-18:
`top -b -n 1` is of no use for the overall cpu and memory usage. Do a proper benchmark.

DragonSerW, 2023-11-18:
7Ji, what command should I enter in terminal to perform test of CPU & GPU usage?

7Ji, 2023-11-18:
https://unix.stackexchange.com/a/414770 shared some useful tools to track usage for a time period and plot the result

7Ji, 2023-11-20:
Chromium-mpp, started with
```
LD_LIBRARY_PATH=/usr/lib/mali-valhall-g610/x11-gbm chromium --use-gl=egl
```
idle with an Arch Wiki page 

CPU and memory usage tracked and plotted with
```
~/.local/bin/procpath record -i 1 -r 120 -d dd.sqlite '$..children[?("chromium" in @.cmdline)]'
~/.local/bin/procpath plot -d ff.sqlite -q cpu -f cpu.svg
~/.local/bin/procpath plot -d dd.sqlite -q rss -f rss.svg
```

1536 is the main executable

[image]

[image]

I don't see a problem

closed 2023-11-20 by 7Ji