7Ji/orangepi5-archlinuxarm #20: No Video Output With mpv ---hwdec=rkmpp <video> with ffmpeg-mpp

JFLim1, 2023-11-20:
Hi @7Ji ,

I don't seems to able to get mpv with vpu hw accelearation working with ffmpeg-mpp.  The video output is a blank blue screen with sound.  It complains:

`[vo/gpu/drmprime] Mapped surface with format 'nv12' has unexpected number of planes. (0 layers and 0 planes, but expected 2 planes)`

Appreciate assistance on this.  Thank you.

```

[jfl@opi5plus ~]$ mpv --hwdec=rkmpp https://www.youtube.com/watch?v=aqz-KE-bpKQ
 (+) Video --vid=1 (*) (vp9 3840x2160 60.000fps)
 (+) Audio --aid=1 (*) (aac 6ch 48000Hz)
File tags:
 Uploader: Blender
 Channel_URL: https://www.youtube.com/channel/UCSMOQeBJ2RAnuFungnQOxLg
Using hardware decoding (rkmpp).
AO: [pipewire] 48000Hz 5.1 6ch floatp
VO: [gpu] 3840x2160 drm_prime[nv12]
[vo/gpu/drmprime] Mapped surface with format 'nv12' has unexpected number of planes. (0 layers and 0 planes, but expected 2 planes)
[vo/gpu/drmprime] mapping DRM dmabuf failed
[vo/gpu] Mapping hardware decoded surface failed.
[vo/gpu/drmprime] Mapped surface with format 'nv12' has unexpected number of planes. (0 layers and 0 planes, but expected 2 planes)
[vo/gpu/drmprime] mapping DRM dmabuf failed
[vo/gpu] Mapping hardware decoded surface failed.
AV: 00:00:00 / 00:10:34 (0%) A-V:  0.000 Cache: 0.4s/293KB
```

7Ji, 2023-11-20:
Why do you add `---hwdec=rkmpp`? All players utilizing ffmpeg gets hwdec automatically, without interfering, because rkmpp decoders replace the software decoders as the default decoders.

Forcing that in the application just fks up all the logic.

Just mpv file.

Verify that by write 7 to `/sys/module/rk_vcodec/parameters/mpp_dev_debug` and then watch kernel log with `dmesg -w`

closed 2023-11-20 by 7Ji

JFLim1, 2023-11-20:
> Why do you add `---hwdec=rkmpp`? All players utilizing ffmpeg gets hwdec automatically, without interfering, because rkmpp decoders replace the software decoders as the default decoders.

Thanks for the info. I thought that was necessary to enable "mpv" to enable vpu hw decode and when vpu hw decode is enable mpv will highlight "Using hardware decoding (rkmpp)".

With ffmpeg-mpp and mpv:

> [jfl@opi5plus ~]$ mpv --gpu-context=wayland https://www.youtube.com/watch?v=aqz-KE-bpKQ
>  (+) Video --vid=1 (*) (vp9 3840x2160 60.000fps)
>  (+) Audio --aid=1 (*) (aac 6ch 48000Hz)
> File tags:
>  Uploader: Blender
>  Channel_URL: https://www.youtube.com/channel/UCSMOQeBJ2RAnuFungnQOxLg
> rga_api version 1.9.3_[0]
> AO: [pipewire] 48000Hz 5.1 6ch floatp
> VO: [gpu] 3840x2160 yuv420p
> AV: 00:00:57 / 00:10:34 (9%) A-V:  0.002 Dropped: 1426 Cache: 33s/71MB
Don't think vpu hw decode is being enabled.  The dropped frames is also high.

And example with the same video when vpu hw acceleration is being used.  This is from another distro. 
 The output is:

> jfl@jfl-opi5plus:~$ mpv https://www.youtube.com/watch?v=aqz-KE-bpKQ
>  (+) Video --vid=1 (*) (vp9 3840x2160 60.000fps)
>  (+) Audio --aid=1 (*) (aac 6ch 48000Hz)
> File tags:
>  Uploader: Blender
>  Channel_URL: https://www.youtube.com/channel/UCSMOQeBJ2RAnuFungnQOxLg
> Using hardware decoding (rkmpp).
> AO: [pulse] 48000Hz 5.1 6ch float
> VO: [gpu] 3840x2160 drm_prime[nv12]
> AV: 00:01:14 / 00:10:34 (12%) A-V:  0.000 Dropped: 28 Cache: 55s/109MB
> 
> Exiting... (Quit)
> jfl@jfl-opi5plus:

JFLim1, 2023-11-20:
Hi @7Ji,

dmesg has these failures:

[    9.258343] [drm] failed to init overlay plane Cluster0-win1
[    9.258345] [drm] failed to init overlay plane Cluster1-win1
[    9.258348] [drm] failed to init overlay plane Cluster2-win1
[    9.258350] [drm] failed to init overlay plane Cluster3-win1

basically with "mpv <video>" are render without vpu hw acceleration.

With "mpv --hwdec=rkmpp <video>" mpv tried to play with vpu hw acceleration but without Vidoe Output just Blank Blue Screen with out.

7Ji, 2023-11-20:
> Just mpv file.