7Ji/orangepi5-archlinuxarm #22: 5b-nightly image kernel panics on boot and locks up

DaveWK, 2023-12-01:
I flashed the image to an SD card, and after a few reboots, the system kernel panics and locks up. Logs attached.

[arch_opi5b_crash.log](https://github.com/7Ji/orangepi5-archlinuxarm/files/13527955/arch_opi5b_crash.log)

7Ji, 2023-12-01:
> after a few reboots

So this is not the case for a new image? What did you do among these "a few reboots"?

I see there's pcie linking traces, was that m.2 disk always connected?

And since you've asked on rkloader repo, what's the bootloader you're using? The mainline one currently is not stable and has caused a few issues.

And I have some minor doubts that my kernel package could've been broken... I've forced a rebuild for the kernel and let's see if it really is the kernel's issue.

7Ji, 2023-12-02:
FYI I've forced rebuild for all kernel packages on the repo. But as I only have a single OPI5 in my hand for the building work that would take more than two days before new packages are pushed. And I still won't be able to test those packages as the builder couldn't be taken offline without breaking the repo update.

7Ji, 2023-12-03:
@DaveWK The kernels are now rebuilt. Please either force an update of the kernel package, or flash the latest nightly.

7Ji, 2023-12-08:
If that does not fix the problem then I would need to force an older version of DDR firmware for the rkloader.

I have that guess because even OPi themselves reported to me that their latest firmware is broken on my OPi5 Plus (I sent back for repair after I had a similar problem on OPi5 Plus, and tested to confirm that problem also appears on their official images), but their older images do not.

7Ji, 2023-12-08:
I've forced the rkloader repo to pack `v1.11` ddr firmware, and this does fix the problem on my 5 Plus:
https://github.com/7Ji/orangepi5-rkloader/commit/2b6ba35faa02ca0138551070a5a29ec4a63bc0a1

I've also forced a rebuild of images in this repo. Please test the latest nightly and check if they fix your problem.

closed 2023-12-09 by 7Ji