7Ji/orangepi5-archlinuxarm #24: locale error

wyf9661, 2023-12-12:
I know it looks like stupid, but when I use the latest image of opi5plus(Dec 12), when I edit `/etc/locale.gen` , add `zh_CN.UTF-8` and then edit `/etc/locale.conf` from  `en_US.UTF-8` to  `zh_CN.UTF-8` .when I install plasma and start sddm, the system language is still English and I click setting icon and change  language from English to Chinese Simplify, The system is half Chinese and half English.
[image]

[image]

[image]

When I use `vim` to open a text with Chinese character,  all Chinese characters are Garbled codes.

But when I roll back the image to Nov. 12, after I change  `/etc/locale.conf` and start sddm, it is already a DE with Chinese LANG. I think there must be someting different on locale which I don't know.
[image]

7Ji, 2023-12-12:
On Nov.19 there was a rootless rewrite which completely changed the building routine. I would guess some configs, ownership and permissions of files were changed without explicitly written as so. But I did not use either of the versions for headful use cases, nor could I test for each release. I only did headless booting and simple tests after the rewrite and did not test around GUI stuffs.

I could only do proper test ~2 weeks later from now, so I'm sorry I can't solve the issue for now.

wyf9661, 2023-12-12:
OK, I do not have a large-capacity SD card or any other things which can use now, I will buy one emmc and test it later.
Or maybe you could test this way if you have some installation media first if it does not waste too much time. I'm sincerely sorry if this bothered you.
Thank you~

7Ji, 2023-12-15:
Can you check if your old local image still work if updated (pacman -Syu)?

Because it seems the latest KDE update was broken and a lot of Arch (non ALARM) users were affected.

https://bbs.archlinuxcn.org/viewtopic.php?pid=58064#p58064

(An example, but they're also a lot of similar reports in #archlinuxcn telegram group)

wyf9661, 2023-12-15:
On both situation, all packages I've installed are all latest.
[image]

wyf9661, 2023-12-15:
[image]

wyf9661, 2023-12-15:
Oh, yes, what he said is the same print with my terminal's output. when I install plasma, the version of the components I think can not be different. It dues to the time it was installed, not the image itself, But I don't know why the situation is different if the version is same.   
[image]

7Ji, 2023-12-15:
You can get a sorted list of installed packages by `pacman -Q` and compare the two.

wyf9661, 2023-12-15:
But too many components. All right, I will download the latest image and have another try.
thanks a lot~

7Ji, 2023-12-15:
You know you could run `diff` or other text comparison tools on the results right?

wyf9661, 2023-12-15:
[image]

wyf9661, 2023-12-15:
I use `yay -Ss plasma > 1.txt` and get two totally same text, and this time everything is ok. I think maybe it is really different kde components that cause the difference. I will close the issue, thank you very much.

wyf9661, 2023-12-15:
Buy the way, I find that boot partition in this image is 256M, much larger than image in Nov.12, which is only 100M, In the version of Nov 12, I cannot install two kernel package because the space is not enough, but now I think I can have a try, thanks again for enlarge the space.

closed 2023-12-15 by wyf9661

7Ji, 2023-12-15:
Doing `yay -Ss plasma` is of no use: that searches only the remote db, not your local one, and the remote db is of couse the same regardless of your local installation. What you should compare are the local installed package lists, gotten with `pacman -Q`

7Ji, 2023-12-15:
> but now I think I can have a try

You don't need to, both of the kernels came pre-installed, and to switch the kernel you only need to change `DEFAULT` in `/boot/extlinux/extlinux.conf`