7Ji/orangepi5-archlinuxarm #25: Orange PI 5 Plus - no login prompt

alex-morgun, 2023-12-25:
Hello,

I copied system to emmc and it boots, I see dmesg log, addingand removing usb devices updates this log.

Login prompt does not appear, what could be wrong?

alex-morgun, 2023-12-25:
Also network interface did not get up, no dhcp requests seen.

7Ji, 2023-12-25:
> copied system to emmc

How? If you extracted the -root.tar release you would also need to change the DTB, as the default DTB is only for 5, not for 5B or 5 Plus

alex-morgun, 2023-12-25:
I installed the [ArchLinuxARM-aarch64-OrangePi5-20231225_022238-rkloader-orangepi_5_plus.img.gz](https://github.com/7Ji/orangepi5-archlinuxarm/releases/download/nightly/ArchLinuxARM-aarch64-OrangePi5-20231225_022238-rkloader-orangepi_5_plus.img.gz)

Which I expected to be a version for Orange PI 5 Plus. Is it correct?

7Ji, 2023-12-25:
Yes it is, but how could/did you
> copied system

From an image to eMMC?

Did it work on SD?

alex-morgun, 2023-12-25:
I copied .img file to SD, and used dd to copy it to the eMMC

alex-morgun, 2023-12-25:
I just tried to image the CD card and it boots this way.

7Ji, 2023-12-25:
I just flashed the latest image onto my 5 Plus after I completely erased all data on it and the image worked just fine:
```
Arch Linux 5.10.110-6-aarch64-orangepi5+ (ttyFIQ0)

alarm login: alarm
Password: 
[   25.654444] systemd-journald[351]: /var/log/journal/6d23dc3cc463415d8dd47be37ebd97a9/user-1000.journal: Realtime clock jumped backwards relative to last journal entry, rotating.
Last login: Wed Dec 20 15:45:43 on ttyFIQ0
[alarm@alarm ~]$ ip[   41.261893] vcc_mipicsi0: disabling
[   41.261949] vcc_mipicsi1: disabling
[   41.261968] vcc_mipicsi1: disabling
 l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: enP3p49s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether c0:74:2b:ff:11:14 brd ff:ff:ff:ff:ff:ff
3: enP4p65s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether c0:74:2b:ff:11:15 brd ff:ff:ff:ff:ff:ff
[alarm@alarm ~]$
```

I can't think of any reason how your board would work differently

closed 2023-12-27 by 7Ji