7Ji/orangepi5-archlinuxarm #26: kernel 6.1

wyf9661, 2023-12-27:
I bump the kernel into v6.1 , using PKGBUILD with [https://github.com/7Ji-PKGBUILDs/linux-panthor](https://github.com/7Ji-PKGBUILDs/linux-panthor), changing branch from `panthor+mpp+rga` to `linux-6.1-rkbsp6-panthor-v4`, and restart my PC. Still use panfork driver, it runs more smoothly with fewer issue, will this pkg be added into arch repo?
 
[image]

wyf9661, 2023-12-27:
I still try to install panthor dirver, but the mesa vendor is still recognized to `llvmpipe`, but it does not matter, now I am really excited to use new kernel, haha!

7Ji, 2023-12-27:
6.7-rc7 was already tagged on Dec 24, 3 days ago. And if there's no rc8 then 6.7 would arrive just a couple days from now. No need to hurry and fiddle around the release candidate kernel. My plan is to enable support for rk3588 in https://github.com/7Ji-PKGBUILDs/linux-aarch64-7ji after 6.7 is released.

wyf9661, 2023-12-27:
Someones says that the VOP will not be merged in 6.7, does it matter?

7Ji, 2023-12-27:
Sorry I didn't read the issue title and thought this was the continuation of another issue.

Tracking issue for that kernel package and mesa-panthor is at https://github.com/7Ji-PKGBUILDs/.report/issues/3 . They were created by hbiyik (who maintains the fork kernel) and back then they were considered not ready for public use. You can continue the discussion there and ping him for in-depth info.

About VOP, I didn't check the info. But whether merged or not I'll add the rk3588 support anyway, as I got more and more tired to have distinct kernel packages among all my devices.

wyf9661, 2023-12-27:
thanks a lot

closed 2023-12-27 by wyf9661

7Ji, 2023-12-31:
Seems like there's rc8: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=f016f7547aeedefed9450499d002ba983b8fce15

So we're gonna wait for an extra week before 6.8

wyf9661, 2023-12-31:
Happy mew year!
will `linux-aarch64-7ji` support `panthor` in 6.8？

7Ji, 2024-01-07:
> will `linux-aarch64-7ji` support `panthor` in 6.8？

If upstream supports it then the package would support it; if upstream doesn't, then no support as I don't want to introduce another big chunky patchset (I already have [one](https://github.com/7Ji-PKGBUILDs/linux-aarch64-7ji/releases/download/assets/sha256-e138ce7b90cd22226e889f88dfd4973d851f7a8f05f667e28cbd4883ba4961dc-0001-rebase-local-changes-to-v6.5.2.patch.xz) for the Amlogic platform and I decided no more local patches).

wyf9661, 2024-01-10:
> @wyf9661
> 
> Could you please share your mainline kernel log? I've bumped -7ji to 6.7 but the mmc module seems to be broken. I want to check out which mmc driver the mainline kernel is using and should be using.
> 
> [7Ji-PKGBUILDs/.report#12](https://githubfast.com/7Ji-PKGBUILDs/.report/issues/12)

I will bump to v6.7 and have a try soon.