7Ji/orangepi5-archlinuxarm #28: This is the best performing distro!

alex-morgun, 2024-01-03:
I tested ubuntu-rockchip and orangepi-archlinuxarm performs the best! It is the best performing distro for RK3588 at the moment. Also it consumes 25% less energy.
 
https://github.com/Joshua-Riek/ubuntu-rockchip

Thanks!

alex-morgun, 2024-01-03:
https://browserbench.org/Speedometer2.1/

AMD 5600g - 153
OrangePi Plus - 75
Samsug S20fe - 43

7Ji, 2024-01-04:
This is not a distro in any sense. It builds upon the unmodified Arch Linux ARM base, the only thing added is my repo (7Ji/archrepo) and some packages from my repo. You can pacstrap a new installation of ALARM and add my repo/packages and it'll work the same.

About speed, it's probably because that I don't do any local config nor install any other packages other than the bare minimum to make the image bootable. I don't like predefined configs for any thing I use, so there goes "user-friendly" but comes max customization.

closed 2024-01-05 by 7Ji