7Ji/orangepi5-archlinuxarm #29: After iwd update, iwd stopped working

alex-morgun, 2024-01-18:
Hello!

- iwctl fails to find an AP after scan
- can't connect wifi
- new pkg iwd-2.13-2

alex-morgun, 2024-01-19:
I downgraded iwd and it works as expected

7Ji, 2024-01-19:
Then it's an ALARM issue? I guess there's nothing I could do here.

7Ji, 2024-01-19:
```
nomad7ji@hk1 /e/s/network> ip address show dev wlan0
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq state UP group default qlen 1000
    link/ether d0:22:be:ea:1a:f1 brd ff:ff:ff:ff:ff:ff
    inet 192.168.7.217/24 metric 1024 brd 192.168.7.255 scope global dynamic wlan0
       valid_lft 865sec preferred_lft 865sec
    inet6 fdb5:c701:19a6:0:d222:beff:feea:1af1/64 scope global mngtmpaddr noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 2408:8214:1218:2610:d222:beff:feea:1af1/64 scope global dynamic mngtmpaddr noprefixroute 
       valid_lft 204078sec preferred_lft 117678sec
    inet6 fe80::d222:beff:feea:1af1/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
nomad7ji@hk1 /e/s/network> pacman -Qs iwd
local/iwd 2.13-2
    Internet Wireless Daemon
```
Works without any issue

closed 2024-01-19 by 7Ji