7Ji/orangepi5-archlinuxarm #30: Error occurs when I install mesa-panfork-git and mali-valhall-g610-firmware

misaki-sxc, 2024-01-26:
Hi，
when I excute 

`pacman -Syu mesa-panfork-git mali-valhall-g610-firmware
`
everything is fine, but when the process goes to (2/2) Updating linux initcpios..., error occurs. here is the whole outputs.

`
(1/2) Arming ConditionNeedsUpdate...
(2/2) Updating linux initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux-aarch64-orangepi5-git.preset: 'default'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-aarch64-orangepi5-git -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-aarch64-orangepi5-git.img
==> Starting build: '5.10.160-r1080192-695850f9bde2-ced0156-1-aarch64-orangepi5-git+'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: '/boot/initramfs-linux-aarch64-orangepi5-git.img'
==> WARNING: errors were encountered during the build. The image may not be complete.
==> Building image from preset: /etc/mkinitcpio.d/linux-aarch64-orangepi5-git.preset: 'fallback'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-aarch64-orangepi5-git -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-aarch64-orangepi5-git-fallback.img -S autodetect
==> Starting build: '5.10.160-r1080192-695850f9bde2-ced0156-1-aarch64-orangepi5-git+'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
==> WARNING: Possibly missing firmware for module: 'bfa'
==> WARNING: Possibly missing firmware for module: 'qla1280'
==> WARNING: Possibly missing firmware for module: 'cxgb4'
==> WARNING: Possibly missing firmware for module: 'csiostor'
==> WARNING: Possibly missing firmware for module: 'cxgb3'
==> WARNING: Possibly missing firmware for module: 'wd719x'
==> WARNING: Possibly missing firmware for module: 'aic94xx'
==> WARNING: Possibly missing firmware for module: 'qla2xxx'
==> WARNING: Possibly missing firmware for module: 'advansys'
==> WARNING: Possibly missing firmware for module: 'ums_eneub6250'
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: '/boot/initramfs-linux-aarch64-orangepi5-git-fallback.img'
==> WARNING: errors were encountered during the build. The image may not be complete.
==> Building image from preset: /etc/mkinitcpio.d/linux-aarch64-orangepi5.preset: 'default'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-aarch64-orangepi5 -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-aarch64-orangepi5.img
==> Starting build: '5.10.110-6-aarch64-orangepi5+'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: '/boot/initramfs-linux-aarch64-orangepi5.img'
==> WARNING: errors were encountered during the build. The image may not be complete.
==> Building image from preset: /etc/mkinitcpio.d/linux-aarch64-orangepi5.preset: 'fallback'
==> Using configuration file: '/etc/mkinitcpio.conf'
  -> -k /boot/vmlinuz-linux-aarch64-orangepi5 -c /etc/mkinitcpio.conf -g /boot/initramfs-linux-aarch64-orangepi5-fallback.img -S autodetect
==> Starting build: '5.10.110-6-aarch64-orangepi5+'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
==> WARNING: Possibly missing firmware for module: 'bfa'
==> WARNING: Possibly missing firmware for module: 'qla1280'
==> WARNING: Possibly missing firmware for module: 'cxgb4'
==> WARNING: Possibly missing firmware for module: 'csiostor'
==> WARNING: Possibly missing firmware for module: 'cxgb3'
==> WARNING: Possibly missing firmware for module: 'wd719x'
==> WARNING: Possibly missing firmware for module: 'aic94xx'
==> WARNING: Possibly missing firmware for module: 'qla2xxx'
==> WARNING: Possibly missing firmware for module: 'advansys'
==> WARNING: Possibly missing firmware for module: 'ums_eneub6250'
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: '/boot/initramfs-linux-aarch64-orangepi5-fallback.img'
==> WARNING: errors were encountered during the build. The image may not be complete.
error: command failed to execute correctly
`

I have search the archlinux forums, they give similar situation, but the problem can't be resolved when I their solutions. URL: https://bbs.archlinuxcn.org/viewtopic.php?id=13563

thank you for your help.

misaki-sxc, 2024-01-26:
Additionally, when I excute mkinitcpio, everything is OK.

```shell

==> Starting dry run: '5.10.160-r1080192-695850f9bde2-ced0156-1-aarch64-orangepi5-git+'
  -> Running build hook: [base]
  -> Running build hook: [udev]
  -> Running build hook: [autodetect]
  -> Running build hook: [modconf]
  -> Running build hook: [kms]
  -> Running build hook: [keyboard]
  -> Running build hook: [keymap]
  -> Running build hook: [consolefont]
==> WARNING: consolefont: no font found in configuration
  -> Running build hook: [block]
  -> Running build hook: [filesystems]
  -> Running build hook: [fsck]
==> Generating module dependencies
==> Dry run complete, use -g IMAGE to generate a real image

```

7Ji, 2024-01-26:
Disable the kms hook in `/etc/mkinitcpio.conf`, or you can just ignore that "error". It happens because mkinitcpio expects to find at least one kernel mode setting module, for x86_64 it's an sane assumption, but on aarch64 this is not the case.

misaki-sxc, 2024-01-26:
Thank you. After disable kms hook, it works.

closed 2024-01-26 by misaki-sxc