7Ji/orangepi5-archlinuxarm #34: Hello 7Ji, how can I enable KVM.

misaki-sxc, 2024-03-26:
Recently, I'm leaning xv6 operating system, It only supports x86 or riscv, so I'm planning using qemu-system-x86_64 + kvm to emulate its environment on my orangepi 5 plus. But I find that the kernel does not support virtualization. I have tried some commands to check it.
```
lscpu | grep Virtualization
grep -E --color=auto 'vmx|svm|0xc0f' /proc/cpuinfo
lsmod | grep kvm
```
But I don't get any output. I am not familiar with the kernel compilation, Do I need to recompile the kernel to enable it？
Please give me some instructions, thank you.

7Ji, 2024-03-26:
It would not be really feasible becuase:
 - KVM is completely broken on Android downstream kernels, and Rockchip's 5.10 & 6.1 vendor kernels for RK3588 based on Android therefore won't work correctly with KVM.
 - Packaging of `qemu-user-` packages are currently broken on Arch Linux ARM and they're therefore not available in ALARM repo. You could install `qemu-system-` but QEMU's functionality is therefore far from complete.

And I don't think KVM is really needed here, as xv6 doesn't really need it to run. Even on x86 and risc-v, xv6 works without KVM, or any other sort of para-virtualization. Why do you think you would need KVM to run it?

misaki-sxc, 2024-03-26:
Thank you for your reply. I know xv6 is a very tiny system, It doesn't need virtualization. But out of curiosity I also tried ubuntu and it took me five minutes to start.

closed 2024-03-26 by misaki-sxc