7Ji/orangepi5-archlinuxarm #35: The problem with the desktop after the update

Vyacheslav-S, 2024-04-18:
Hi!  
Yesterday, several packages from the repository [7Ji repo](https://github.com/7Ji/archrepo) were updated and there was a problem with the start of gnome-session.
dmesg after booting the system - https://pastebin.com/07TiUCkL
Now I can only log in to the console tty2 ( `ctrl + alt + f2` ).
I deleted `mesa-panfork-git` and installed the regular `mesa` package.
After the reboot, I can now log in normally. But now there is no GPU/ 3D Acceleration.
Any help you can give me, please.

7Ji, 2024-04-18:
The updated `mali-valhall-g610-firmware` package was proven to be troublesome and has been reverted in https://github.com/7Ji-PKGBUILDs/mali-valhall-g610-firmware/commit/7edcc2b77084bd4c950399abcefa19dd3f71fbf5 . Could you try to install the newly released, reverted package?

Vyacheslav-S, 2024-04-18:
Yesterday I updated the system and the version of the mali-valhall-g610-firmware package became g21p0.
After a few hours, I rebooted the system and the desktop became unavailable.
Then I logged into the tty2 console and checked for updates.
A message appeared that the system has a newer package mali-valhall-g610-firmware with version g21p0, and the repository version is g18p0.
OK, I've downgraded the version to g18p0.
But after the reboot, I was unable to log in to the desktop. The dmesg errors were the same.
Then I replaced mesa-panfork-git with a regular mesa package in the console.
I rebooted the system and the desktop appeared, but there is no acceleration.

Today I restored the system from a saved working image of Clonezilla, which I did a month ago.
Everything worked fine. Next, I updated all the packages again. And again, the desktop is unavailable and the same errors in dmesg.
Maybe the problem is in another package?

7Ji, 2024-04-19:
That package was not reverted correctly: https://github.com/7Ji-PKGBUILDs/mali-valhall-g610-firmware/commit/7edcc2b77084bd4c950399abcefa19dd3f71fbf5#commitcomment-141125662

The correctly reverted firmware should be released after a few hours

Vyacheslav-S, 2024-04-19:
>  The correctly reverted firmware should be released after a few hours

@7Ji , Thanks! 
Now the desktop is working, the problem is solved.
```
alarm@opi5p:~$ pacman -Qs mali-valhall-g610-firmware
local/mali-valhall-g610-firmware g18p0.01eac0-4
    Firmware for ARM Mali-G610
```

closed 2024-04-19 by 7Ji