7Ji/orangepi5-archlinuxarm #36: Enable Discussion

winksaville, 2024-04-28:
@7Ji, I'd like to suggest enabling [Discussions](https://docs.github.com/en/discussions/quickstart) in the repo's settings so noobs like me can ask questions and you won't have to repeat yourself as much :)

(My question is going to be "How does your Arch distro compare to [Orange's distro](http://www.orangepi.org/html/softWare/orangePiOS/arch.html))?"

7Ji, 2024-04-28:
This is not my distro. This is vanilla Arch Linux ARM with https://github.com/7Ji/archrepo added and some packages from there pre-installed.

closed 2024-04-29 by 7Ji

7Ji, 2024-04-29:
> (My question is going to be "How does your Arch distro compare to Orange's distro)?"

https://github.com/7Ji/orangepi5-archlinuxarm?tab=readme-ov-file#prebuilt-archlinux-arm-images-for-orange-pi-5--5b--5-plus

Isn't that obvious that this IS Arch Linux ARM, through and through, not yet another distro fork?