7Ji/orangepi5-archlinuxarm #38: Installation of packages failed due to the error "bsdtar: Failed to set default locale".

hronro, 2024-05-07:
I haven't installed any DE yet, I'm currently using SSH from my MacBook.

7Ji, 2024-05-07:
Two things:
- `bsdtar: Failed to set default locale` is not a fatal error, it should not prevent `mkinitcpio` from being finished
- `mkinitcpio` on ALARM is pretty much broken now, I'd recommend to use `booster` as the initcpio generator, and I've set that as default after https://github.com/7Ji/orangepi5-archlinuxarm/commit/a0be71252ec44e345df02095dfd8193a1e39ed36

closed 2024-05-09 by 7Ji

hronro, 2024-05-17:
I apologize for reaching out here. I have reinstalled the operating system using the latest 5_plus.img, and everything is functioning properly now.

Although I can execute `pacman -Syu linux-aarch64-7ji` successfully, it seems that the new kernel is not activated after restarting.

Upon inspecting `/boot/extlinux/extlinux.conf`, I noticed that the new kernel is not listed there either.

Do you have any suggestions for me?

7Ji, 2024-05-17:
You need to modify the config by yourself, there'no automcated mechanism to update the extlinux.conf

hronro, 2024-05-17:
Hi @7Ji, thanks for your quick response.

I checked my `/boot` directory and found it seems some files are missing.

[image]

There is only one img file in `/boot`. The `/booster-linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git.img` and the img file for the `linux-aarch64-7ji` kernel are missing.

7Ji, 2024-05-17:
You now only have `linux-aarch64-rockchip-rk3588-bsp5.10-orangepi` installed, the others are not "missing", but rather "not removed" when you uninstall the corresponding kernel. Install them again if you want to use them.

hronro, 2024-05-17:
I didn't uninstall `linux-aarch64-7ji`.

7Ji, 2024-05-17:
Then run `/usr/lib/booster/regenerate_images` to regenerate all booster images