7Ji/orangepi5-archlinuxarm #41: GPU Acceleration

jackrosenthal, 2024-06-02:
Hi,

I just set up my OrangePi 5+ with this via `pacstrap`.  Upon installing a desktop environment (X11+i3), and `mesa-panfork-git mali-valhall-g610-firmware`, I'm not getting any hardware acceleration support.

`chromium` without the blob GPU drivers gives a white screen.

With the blob GPU drivers, it appears EGL fails to initialize:

```
arm_release_ver: g13p0-01eac0, rk_so_ver: 10
[2422:2422:0601/210111.347543:ERROR:gl_display.cc(796)] eglInitialize Default failed with error EGL_NOT_INITIALIZED
[2422:2422:0601/210111.347826:ERROR:gl_display.cc(830)] Initialization of all EGL display types failed.
[2422:2422:0601/210111.348377:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[2422:2422:0601/210111.362439:ERROR:gl_display.cc(796)] eglInitialize Default failed with error EGL_NOT_INITIALIZED
[2422:2422:0601/210111.362549:ERROR:gl_display.cc(830)] Initialization of all EGL display types failed.
[2422:2422:0601/210111.362616:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[2422:2422:0601/210111.365092:ERROR:viz_main_impl.cc(166)] Exiting GPU process due to errors during initialization
arm_release_ver: g13p0-01eac0, rk_so_ver: 10
[2522:2522:0601/210111.596998:ERROR:gl_display.cc(796)] eglInitialize Default failed with error EGL_NOT_INITIALIZED
[2522:2522:0601/210111.597153:ERROR:gl_display.cc(830)] Initialization of all EGL display types failed.
[2522:2522:0601/210111.597302:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[2425:2454:0601/210111.612195:ERROR:cookie_monster.cc(1206)] Found 1 duplicate cookies for key='google.com', with {name='receive-cookie-deprecation', domain='www.google.com', path='/'}
[2522:2522:0601/210111.616293:ERROR:gl_display.cc(796)] eglInitialize Default failed with error EGL_NOT_INITIALIZED
[2522:2522:0601/210111.616391:ERROR:gl_display.cc(830)] Initialization of all EGL display types failed.
[2522:2522:0601/210111.616444:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[2522:2522:0601/210111.618671:ERROR:viz_main_impl.cc(166)] Exiting GPU process due to errors during initialization
[2390:2562:0601/210111.643510:ERROR:object_proxy.cc(576)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name is not activatable
[2390:2562:0601/210111.643967:ERROR:object_proxy.cc(576)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name is not activatable
[2390:2562:0601/210111.644359:ERROR:object_proxy.cc(576)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name is not activatable
arm_release_ver: g13p0-01eac0, rk_so_ver: 10
[2575:2575:0601/210111.725000:ERROR:gl_display.cc(796)] eglInitialize Default failed with error EGL_NOT_INITIALIZED
[2575:2575:0601/210111.726140:ERROR:gl_display.cc(830)] Initialization of all EGL display types failed.
[2575:2575:0601/210111.726318:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[2575:2575:0601/210111.738301:ERROR:gl_display.cc(796)] eglInitialize Default failed with error EGL_NOT_INITIALIZED
[2575:2575:0601/210111.738429:ERROR:gl_display.cc(830)] Initialization of all EGL display types failed.
[2575:2575:0601/210111.738501:ERROR:gl_ozone_egl.cc(26)] GLDisplayEGL::Initialize failed.
[2575:2575:0601/210111.753508:ERROR:viz_main_impl.cc(166)] Exiting GPU process due to errors during initialization
[2468:7:0601/210111.974656:ERROR:command_buffer_proxy_impl.cc(131)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
```

`chrome://gpu` has:

```
Graphics Feature Status
=======================
*   Canvas: Software only, hardware acceleration unavailable
*   Canvas out-of-process rasterization: Disabled
*   Direct Rendering Display Compositor: Disabled
*   Compositing: Software only. Hardware acceleration disabled
*   Multiple Raster Threads: Enabled
*   OpenGL: Disabled
*   Rasterization: Software only. Hardware acceleration disabled
*   Raw Draw: Disabled
*   Skia Graphite: Disabled
*   Video Decode: Software only. Hardware acceleration disabled
*   Video Encode: Software only. Hardware acceleration disabled
*   Vulkan: Disabled
*   WebGL: Software only, hardware acceleration unavailable
*   WebGL2: Software only, hardware acceleration unavailable
*   WebGPU: Disabled
*   WebNN: Disabled
```

And observed FPS on WebGL aquarium is ~6 fps.

Other interesting things, Kitty won't run:

```
[0.216] [glfw error 65542]: GLX: GLX extension not found
[0.216] Failed to create GLFW temp window! This usually happens because of old/broken OpenGL drivers. kitty requires working OpenGL 3.1 drivers.
```

What am I missing?  Any other useful info to provide?

7Ji, 2024-06-02:
What's the result of `id`? For mpp to work you need to be in `video` group

jackrosenthal, 2024-06-02:
Yeah I'm a member of the `video` group, and permissions look good too:

```
% id
uid=1000(jrosenth) gid=1000(jrosenth) groups=1000(jrosenth),985(video),998(wheel)
% ls -la /dev/video-*
-rw-rw---- 1 root video 71 Jun  1 20:47 /dev/video-dec0
-rw-rw---- 1 root video  4 Jun  1 20:47 /dev/video-enc0
% ls -la /dev/dri/*
crw-rw----+ 1 root video  226,   0 Jun  1 20:47 /dev/dri/card0
crw-rw----+ 1 root video  226,   1 Jun  1 20:47 /dev/dri/card1
crw-rw-rw-  1 root render 226, 128 Jun  1 20:47 /dev/dri/renderD128
crw-rw-rw-  1 root render 226, 129 Jun  1 20:47 /dev/dri/renderD129

/dev/dri/by-path:
total 0
drwxr-xr-x 2 root root 120 Jun  1 20:47 .
drwxr-xr-x 3 root root 140 Jun  1 20:47 ..
lrwxrwxrwx 1 root root   8 Jun  1 20:47 platform-display-subsystem-card -> ../card0
lrwxrwxrwx 1 root root  13 Jun  1 20:47 platform-display-subsystem-render -> ../renderD128
lrwxrwxrwx 1 root root   8 Jun  1 20:47 platform-fdab0000.npu-card -> ../card1
lrwxrwxrwx 1 root root  13 Jun  1 20:47 platform-fdab0000.npu-render -> ../renderD129
```

jackrosenthal, 2024-06-02:
I'm also attaching the list of installed packages (`pacman -Q`) if that helps.

[installed_packages.txt](https://github.com/user-attachments/files/15525665/installed_packages.txt)

7Ji, 2024-06-03:
> permissions look good too

No those nodes are not what's needed here, mpp hw decoding relies on `/dev/mpp`

> the list of installed packages 

You installed `chromium`, not `chromium-mpp`, it wont' work with mpp. Currently it's not available in repo due to broken builds of newer version. You could, however, build the old version by yourself: https://github.com/7Ji-PKGBUILDs/chromium-mpp/commit/35302ecd541a915b8917a8225bb2aee0059122f1

jackrosenthal, 2024-06-03:
I wasn't able to get chromium-mpp to build at commit 35302ecd541a915b8917a8225bb2aee0059122f1, it appears there's a 404 on an external dependency:

```
jrosenth ~/chromium-mpp % makepkg
==> Making package: chromium-mpp 122.0.6261.128-1 (Sun 02 Jun 2024 09:52:59 PM MDT)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Downloading chromium-122.0.6261.128.tar.xz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 3272M  100 3272M    0     0  56.0M      0  0:00:58  0:00:58 --:--:-- 61.5M
  -> Downloading chromium-launcher-8.tar.gz...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100  4725    0  4725    0     0   7774      0 --:--:-- --:--:-- --:--:--  7774
  -> Downloading chromium-patches-122.tar.bz2...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  5253    0  5253    0     0  14840      0 --:--:-- --:--:-- --:--:-- 14881
  -> Found support-ICU-74-in-LazyTextBreakIterator.patch
  -> Found drop-flag-unsupported-by-clang17.patch
  -> Found compiler-rt-adjust-paths.patch
  -> Found use-oauth2-client-switches-as-default.patch
  -> Found 0001-widevine-support-for-arm.patch
  -> Found 0002-Run-blink-bindings-generation-single-threaded.patch
  -> Found 0003-Fix-eu-strip-build-for-newer-GCC.patch
  -> Found 0004-Optimize-eu-strip-building-logic.patch
  -> Cloning elfutils git repo...
Cloning into bare repository '/home/jrosenth/chromium-mpp/elfutils'...
remote: Enumerating objects: 39054, done.
remote: Counting objects: 100% (39054/39054), done.
remote: Compressing objects: 100% (7687/7687), done.
remote: Total 39054 (delta 32558), reused 37588 (delta 31324), pack-reused 0
Receiving objects: 100% (39054/39054), 21.76 MiB | 25.43 MiB/s, done.
Resolving deltas: 100% (32558/32558), done.
  -> Downloading mpp-0001-HACK-media-Support-V4L2-video-decoder.patch...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404
==> ERROR: Failure while downloading https://github.com/JeffyCN/meta-rockchip/raw/7f01be8b695ed27220c4fb3d92f96f65aeafc755/dynamic-layers/recipes-browser/chromium/chromium_122.0.6261/0001-HACK-media-Support-V4L2-video-decoder.patch
    Aborting...
>>> Command exited with status 1
```

That being said, should `chromium-mpp` be required to get basic performance?  https://github.com/7Ji/orangepi5-archlinuxarm?tab=readme-ov-file#performance-comparison looks like it was done with non-mpp chromium, but I'm barely able to get 6 fps on WebGL aquarium.

7Ji, 2024-06-03:
> looks like it was done with non-mpp chromium

No it's not non-mpp chromium, those are commands, and `chormium-mpp` is only a patched `chromium` also use the `chromium` command

7Ji, 2024-08-06:
Now chromium-mpp works again, but only on BSP 5.10 + X11 Mali blob + Panfork Mesa

See https://github.com/7Ji-PKGBUILDs/.meta/issues/40

closed 2024-08-06 by 7Ji