7Ji/orangepi5-archlinuxarm #43: Prioritize SD card over eMMC

winksaville, 2024-06-21:
I'd like to be able to give SD card a priority if both SD card and eMMC are inserted. Is that is not possible now and if not what would you suggest the best/easiest way to do that?

7Ji, 2024-06-22:
> I'd like to be able to give SD card a priority if both SD card and eMMC are inserted

IIRC that IS the current boot order, i.e. RK3588 is hard-coded to lookup a FIP/**bootloader** (_not bootable media_) from SD -> eMMC -> SPI (for SD or eMMC FIP is stored in the "reserved" area at the head of the drive). The u-boot in FIP could however has its own boot order to look up booting configurations (_not boot entry_)  but u-boot still prefers SD over eMMC in any cases.

I doubt the problem here is that you've written the same image into both SD and eMMC? If that's the case then partitions on SD and eMMC would have same UUIDs, and as root drives are mounted using UUID, it's totally legal that kernel booted from SD mounts the root partition from eMMC. To avoid this, either: a. write different images (e.g. next nightly), or b. use tune2fs to change UUID and update corresponding `/boot/extlinux/extlinux` entry, or c. partition eMMC and install by yourself following the [doc here](https://github.com/7Ji/orangepi5-archlinuxarm?tab=readme-ov-file#manual-installation-using-pacstrap) so you have a clean, dedicated system on eMMC

winksaville, 2024-06-22:
> I doubt the problem here is that you've written the same image into both SD and eMMC?

I did put the same image on both and didn't anticipate that was wrong.

I like option b, is it correct that after using `dd` to copy the image to eMMC I needed to:
 * Update the UUID of the eMMC root partition, `/dev/mmcblk1p4`:
   * [tune2fs](https://linux.die.net/man/8/tune2fs) -U random /dev/mmcblk1p4
 * Edit the extlinux.conf file of the eMMC boot partition, `/dev/mmcblk1p3`
    changing lines 9 and 15 of extlinux.conf:
   * `mount /dev/mmcblk1p3 <path-to-mount-dir>`
   * `vim <path-to-mount-dir>/extlinux/extlinux.conf`

Here is the extlinux.conf on the SD card and eMMC card:
```
[alarm@alarm ~]$ cat -n /boot/extlinux/extlinux.conf 
     1	MENU TITLE Select the kernel to boot
     2	TIMEOUT 30
     3	DEFAULT linux-aarch64-rockchip-rk3588-bsp5.10-orangepi
     4	LABEL	linux-aarch64-rockchip-rk3588-bsp5.10-orangepi
     5		LINUX	/vmlinuz-linux-aarch64-rockchip-rk3588-bsp5.10-orangepi
     6		INITRD	/booster-linux-aarch64-rockchip-rk3588-bsp5.10-orangepi.img
     7		#FDTDIR	/dtbs/linux-aarch64-rockchip-rk3588-bsp5.10-orangepi
     8		FDT	/dtbs/linux-aarch64-rockchip-rk3588-bsp5.10-orangepi/rockchip/rk3588-orangepi-5-plus.dtb
     9		APPEND	root=UUID=08f5ede3-14a0-498f-a9e6-7576f09763c8 rw
    10	LABEL	linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git
    11	LINUX	/vmlinuz-linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git
    12	INITRD	/booster-linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git.img
    13	#FDTDIR	/dtbs/linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git
    14	FDT	/dtbs/linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git/rockchip/rk3588-orangepi-5-plus.dtb
    15	APPEND	root=UUID=08f5ede3-14a0-498f-a9e6-7576f09763c8 rw
[alarm@alarm ~]$ 
```

7Ji, 2024-06-22:
There's one file missing, `/etc/fstab` in root partition also needs to be updated.

Other than that, everything looks OK.

winksaville, 2024-06-22:
> There's one file missing, `/etc/fstab` in root partition also needs to be updated.
> 
> Other than that, everything looks OK.

Got it, txs.

One other question; is there a SD or USB adapter that would allow me to read/write the eMMC card from my laptop or desktop?

7Ji, 2024-06-23:
OPi (or rather, almost any SBC vendor) uses a proprietary connector for their eMMC module, and they have no official adapter (RockPi/Odroid/etc have, but they use different connectors).

But, you can use uboot's ums mode, which emulates the SBC as a usb mass storage device with one of the attached storage device as backing storage. Get a serial cable, interrupt the bootflow to enter uboot shell, type ums enter and you shall get the help message.

winksaville, 2024-06-23:
> OPi (or rather, almost any SBC vendor) uses a proprietary connector for their eMMC module, and they have no official adapter (RockPi/Odroid/etc have, but they use different connectors).

Very unkind and short sighted of them, IMO.

> 
> But, you can use uboot's ums mode, which emulates the SBC as a usb mass storage device with one of the attached storage device as backing storage. Get a serial cable, interrupt the bootflow to enter uboot shell, type ums enter and you shall get the help message.

Txs, have you used that to flash an image, if so what was the read/write transfer rate?

7Ji, 2024-06-23:
I used that, not to flash an image, but to backup. I only do manual installation myself with pacstrap, and images here are only installed and tested on SD after big changes. The speed is capped by USB 2.0 bandwidth, minus the UMS protocol overhead and the USB gadget overhead

winksaville, 2024-06-23:
@7Ji I can now boot from SD card with eMMC inserted and flashed when I change the UUID.

But there was one problem, **sometimes**, when I booted into the SD card it ended up mounting /boot from eMMC.
The problem is that both mmcblk0p3 and mmcblk1p3 have the same "UUID" and they need to be different.
```
`[alarm@alarm ~]$ lsblk -f
NAME         FSTYPE FSVER LABEL     UUID                                 FSAVAIL FSUSE% MOUNTPOINTS
mtdblock0                                                                               
mmcblk1                                                                                 
├─mmcblk1p1                                                                             
├─mmcblk1p2                                                                             
├─mmcblk1p3  vfat   FAT32 ALARMBOOT 6A2D-5CB9                             161.3M    36% /boot
└─mmcblk1p4  ext4   1.0   ALARMROOT 8f808b1d-86b4-4689-8ed0-c66e88ce6a47                
mmcblk1boot0                                                                            
mmcblk0                                                                                 
├─mmcblk0p1                                                                             
├─mmcblk0p2                                                                             
├─mmcblk0p3  vfat   FAT32 ALARMBOOT 6A2D-5CB9                                           
└─mmcblk0p4  ext4   1.0   ALARMROOT 08f5ede3-14a0-498f-a9e6-7576f09763c8  111.2G     6% /
mmcblk1boot1                                                                            
zram0                                                                                   
[alarm@alarm ~]$ 
```

Here I've booted with just eMMC  and installed mtools with `sudo pacman -S mtools` and used
`mlabel` to change the "UUID" to a random value, in this case 286D-FDD8:
```
[alarm@alarm ~]$ sudo mlabel -n -i /dev/mmcblk1p3
[alarm@alarm ~]$ lsblk -f
NAME         FSTYPE FSVER LABEL     UUID                                 FSAVAIL FSUSE% MOUNTPOINTS
mtdblock0                                                                               
mmcblk1                                                                                 
├─mmcblk1p1                                                                             
├─mmcblk1p2                                                                             
├─mmcblk1p3  vfat   FAT32           286D-FDD8                             161.3M    36% /boot
└─mmcblk1p4  ext4   1.0   ALARMROOT 8f808b1d-86b4-4689-8ed0-c66e88ce6a47   26.6G     6% /
mmcblk1boot0                                                                            
mmcblk1boot1                                                                            
zram0                                                                                   
```
And, of course, changed the boot partition UUID to 286D-FDD8 in `/etc/fstab`:
```
     1	# Static information about the filesystems.
     2	# See fstab(5) for details.
     3	
     4	# <file system> <dir> <type> <options> <dump> <pass>
     5	# root partition with ext4 on SDcard / USB drive
     6	UUID=8f808b1d-86b4-4689-8ed0-c66e88ce6a47	/	ext4	rw,noatime	0 1
     7	# boot partition with vfat on SDcard / USB drive
     8	UUID=286D-FDD8	/boot	vfat	rw,noatime	0 2
[alarm@alarm ~]$
```

Now, when I boot to the SD card with the eMMC installed we get the correct /boot:
```
[alarm@alarm ~]$ lsblk -f
NAME         FSTYPE FSVER LABEL     UUID                                 FSAVAIL FSUSE% MOUNTPOINTS
mtdblock0                                                                               
mmcblk1                                                                                 
├─mmcblk1p1                                                                             
├─mmcblk1p2                                                                             
├─mmcblk1p3  vfat   FAT32           286D-FDD8                                           
└─mmcblk1p4  ext4   1.0   ALARMROOT 8f808b1d-86b4-4689-8ed0-c66e88ce6a47                
mmcblk1boot0                                                                            
mmcblk1boot1                                                                            
mmcblk0                                                                                 
├─mmcblk0p1                                                                             
├─mmcblk0p2                                                                             
├─mmcblk0p3  vfat   FAT32 ALARMBOOT 6A2D-5CB9                             161.3M    36% /boot
└─mmcblk0p4  ext4   1.0   ALARMROOT 08f5ede3-14a0-498f-a9e6-7576f09763c8  111.1G     6% /
zram0                                                                                   
[alarm@alarm ~]$
```

So THANK YOU VERY MUCH :)

closed 2024-06-23 by winksaville