7Ji/orangepi5-archlinuxarm #45: Request: Make boot partition bigger and install booster on the OS (kernel patching issues)

DaveWK, 2024-07-03:
Hi,
I am running into a couple of issues when I use this image for an orange pi 5b.

1. When patching the kernel, it seems like booster is not configured or installed on the image, so it doesn't use booster when patching.
2. When patching the kernel, the boot partition constantly runs out of space.

I would propose that the boot partition should be 1 gig by default for a few of reasons.
 1. It's easier to shrink /boot than make it bigger when there's a subsequent partition. If I wanted to make /boot bigger I would need to migrate all the data on / to the new offset and wait for it to copy., then resize /boot
 2. Since it's empty space it should compress well.
 3. the image as it stands now cannot update the kernel because it runs out of space during initramfs

which brings up my second ask, if you could kindly set up booster in the image so it works when patching the same way as when the image was created.

7Ji, 2024-07-12:
> When patching the kernel, it seems like booster is not configured or installed on the image, so it doesn't use booster when patching.

Did you install-after-uninstall or uninstall-after-install? If you uninstall the existing kernels with `-Rcns` then booster would also be uninstalled as pacman would consider it a dependency no longer needed. Do with plain `-R` instead. Or if you do insist on doing so, re-install booster manually before install new kernels, so they won't grab `mkinitcpio` as the default `initramfs` provider.

> When patching the kernel, the boot partition constantly runs out of space.

What do you mean by "patching the kernel"? Things break in the stage when kernel image is installed or in the stage when initcpio image is being generated?

I don't want to adjust the partition layout yet: if you want a custom partition layout you can always use the `-root.tar` release, and you can partition your SD card in whatever way your want. The current image is partitioned with maximum space utilization in mind so it would: 1) fit on a 4G SD card 2) encourages users to resize root partition if they really want to use it 3) encourage users to actually do an Arch install with either the `root.tar` release here or `pacstrap`.

7Ji, 2024-07-15:
Closed by https://github.com/7Ji/orangepi5-archlinuxarm/commit/62efd484089cf0918ce2ba8ad8171c6bd01fb915

Now we have four kernels pre-installed:
- `linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git`, is kept and replaced the removed linux-aarch64-rockchip-rk3588-bsp5.10-orangepi as the default kernel
- `linux-aarch64-rockchip-bsp6.1-joshua-git` as recommended kernel
- `linux-aarch64-rockchip-armbian-git` as the alternative cutting edge try-it-all
- `linux-aarch64-7ji` as the stable mainline

Now boot partition size is 512M, and we still have 234M free. 1G is way too much overkill

Now whole image size is 3G, it should still fit on a 4G SD card.

closed 2024-07-15 by 7Ji