7Ji/orangepi5-archlinuxarm #51: Installing kernel packages from an existing archlinux arm install?

jcelerier, 2024-08-07:
I have an Orange Pi 5+ running archlinux arm with 5.10 kernel.

I would like to use the 6.1 kernel sources provided here without reinstalling everything, how would I got about it ?

- Can this kernel be rebuilt easily as a PKGBUILD (I need some additional kernel patches for a sound card I'm using) ? e.g. can I just `makepkg -si` and things will go to the right place ?
- Will it replace the kernel in an existing Arch Linux ARM install without issues ?
- Will it get me GPU acceleration in any OpenGL-using app ? 
Thanks !

7Ji, 2024-08-07:
All kernel packages at https://github.com/7Ji-PKGBUILDs can co-exist with each other, as they store their files in unique places similar to Arch on x86, unlike official ALARM kernels. So:
1. You can install a 6.1 kernel alongside the old 5.10
2. Without extlinux.conf changes you would still boot into the old 5.10 kernel, nothing would be broken.
3. You can add your patches to another, your own kernel package that uses yet another name so it co-exist with the original 6.1
4. You're recommended to add the new kernel as a new extlinux.conf label and set it as new default, keeping the old one, so it would be easy to revert back to the old kernel to fix things up.

On 6.1, it's still not entirely natively possible to do OpenGL. The backported panthor is an OpenGLES-only driver (paired with userspace mesa-panfrost-git) with experimental Vulkan support (if paired with mesa-panvk-git). The hardware itself was no capable of non-ES OpenGL after all. But most Linux apps should work with ES. Even if you really want to run GL-only apps, you could install gl4es-git, which maps GL calls to GLES calls so the hardware could do it.

jcelerier, 2024-08-07:
Great. I can work with OpenGL ES or Vulkan, thanks !

jcelerier, 2024-08-07:
Would https://github.com/7Ji-PKGBUILDs/linux-aarch64-7ji/ work on a Orange Pi 5+ ? or do I need https://github.com/7Ji-PKGBUILDs/linux-aarch64-rockchip-bsp6.1-joshua-git ?

jcelerier, 2024-08-07:
I've modified extlinux.conf this way: 

```
default orangepi
menu title orange pi
prompt 1
timeout 50

LABEL orangepi
    MENU LABEL Primary kernel
    LINUX /vmlinuz-linux-aarch64-orangepi5
    FDT /dtbs/linux-aarch64-orangepi5/rockchip/rk3588-orangepi-5-plus.dtb
    APPEND initrd=/initramfs-linux-aarch64-orangepi5.img console=ttyS2,1500000 root=PARTUUID=0aa270b8-8069-4729-bf1c-38f0c546e2a8 rw rootwait audit=0 plymouth.ignore-serial-consoles loglevel=2 cma=128M mitigations=off threadirqs power_save=0 preempt=full usbcore.autosuspend=-1 blacklist=snd-usb-audio

LABEL test
    MENU LABEL 6.1 kernel
    LINUX /vmlinuz-linux-aarch64-rockchip-bsp6.1-joshua-git
    FDT /dtbs/linux-aarch64-orangepi5/rockchip/rk3588-orangepi-5-plus.dtb
    APPEND initrd=/initramfs-linux-aarch64-rockchip-bsp6.1-joshua-git.img console=ttyS2,1500000 root=PARTUUID=0aa270b8-8069-4729-bf1c-38f0c546e2a8 rw rootwait audit=0 plymouth.ignore-serial-consoles loglevel=2 cma=128M mitigations=off threadirqs power_save=0 preempt=full usbcore.autosuspend=-1 blacklist=snd-usb-audio
```

but I don't manage to get a boot menu to show up. If I set the default to "test" which is the new kernel the Orange Pi does not boot at all. What am I missing ? thanks!

jcelerier, 2024-08-07:
I managed to boot the 5.10 kernel provided by 7Ji-PKGBUILDs. But the 6.1 and 6.10 don't boot.

7Ji, 2024-08-08:
Two things:
1. The menu would only show on serial. There's no HDMI video support currently enabled in u-boot. You need a USB-serial cable.
2. Each kerne needs their own initcpio, defined at option `INITRD`

As an example, here's my current extlinux.conf on my opi5:
```
MENU TITLE Select a kernel to boot
TIMEOUT 10
DEFAULT linux-aarch64-rockchip-bsp6.1-joshua-git
LABEL   linux-aarch64-rockchip-bsp6.1-joshua-git
        LINUX   /vmlinuz-linux-aarch64-rockchip-bsp6.1-joshua-git
        INITRD  /booster-linux-aarch64-rockchip-bsp6.1-joshua-git.img
        FDTDIR  /dtbs/linux-aarch64-rockchip-bsp6.1-joshua-git
        APPEND  root=/dev/nvme0n1p2 rootflags=subvol=@,compress=zstd:3 rw resume=/dev/nvme0n1p3
LABEL   linux-aarch64-rockchip-bsp5.10-joshua-git
        LINUX   /vmlinuz-linux-aarch64-rockchip-bsp5.10-joshua-git
        INITRD  /booster-linux-aarch64-rockchip-bsp5.10-joshua-git.img
        FDTDIR  /dtbs/linux-aarch64-rockchip-bsp5.10-joshua-git
        APPEND  root=/dev/nvme0n1p2 rootflags=subvol=@,compress=zstd:3 rw resume=/dev/nvme0n1p3
LABEL   linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git
        LINUX   /vmlinuz-linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git
        INITRD  /booster-linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git.img
        FDTDIR  /dtbs/linux-aarch64-rockchip-rk3588-bsp5.10-orangepi-git
        APPEND  root=/dev/nvme0n1p2 rootflags=subvol=@,compress=zstd:3 rw resume=/dev/nvme0n1p3
```

closed 2024-08-08 by jcelerier

7Ji, 2024-08-09:
A thing I noticed: you didn't named your modified as a different kernel from that's available in [my repo](https://github.com/7Ji/archrepo). If you have my repo configured in your pacman.conf, then future updates would probably replace it.