7Ji/orangepi5-archlinuxarm #52: Mainline Mesa from official Arch Linux ARM repo is working on Orange Pi 5 Plus with vendor kernel 6.1!

ivanbrash, 2024-08-11:
Logs:
`[alarm@archlinux ~]$ glxinfo -B
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Mesa (0xffffffff)
    Device: Mali-G610 (Panfrost) (0xffffffff)
    Version: 24.1.5
    Accelerated: yes
    Video memory: 15961MB
    Unified memory: yes
    Preferred profile: core (0x1)
    Max core profile version: 3.1
    Max compat profile version: 3.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: Mesa
OpenGL renderer string: Mali-G610 (Panfrost)
OpenGL core profile version string: 3.1 Mesa 24.1.5-arch1.1
OpenGL core profile shading language version string: 1.40
OpenGL core profile context flags: (none)

OpenGL version string: 3.1 Mesa 24.1.5-arch1.1
OpenGL shading language version string: 1.40
OpenGL context flags: (none)

OpenGL ES profile version string: OpenGL ES 3.1 Mesa 24.1.5-arch1.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10

[alarm@archlinux ~]$ 
`
Screenshots:
[image]

7Ji, 2024-08-12:
Good news! We can finally drop `mesa-panfrost-git` from the repo.

7Ji, 2024-08-12:
So I did a simple test with `glmark2-es-wayland` on both `mesa` (mainline) and `mesa-panfrost-git` (the merge source), the mainline, able to run on G610 now, performs pretty much the same as `mesa-panfrost-git`, as expected.

Both running `glmark2-es-wayland --size 3840x2160 --off-screen`:

With `mesa-mainline-24.1.5-arch1.1`:
```
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      Mesa
    GL_RENDERER:    Mali-G610 (Panfrost)
    GL_VERSION:     OpenGL ES 3.1 Mesa 24.1.5-arch1.1
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   3840x2160 windowed
=======================================================
[build] use-vbo=false: FPS: 524 FrameTime: 1.909 ms
[build] use-vbo=true: FPS: 577 FrameTime: 1.734 ms
[texture] texture-filter=nearest: FPS: 466 FrameTime: 2.148 ms
[texture] texture-filter=linear: FPS: 464 FrameTime: 2.159 ms
[texture] texture-filter=mipmap: FPS: 469 FrameTime: 2.134 ms
[shading] shading=gouraud: FPS: 483 FrameTime: 2.072 ms
[shading] shading=blinn-phong-inf: FPS: 466 FrameTime: 2.148 ms
[shading] shading=phong: FPS: 365 FrameTime: 2.746 ms
[shading] shading=cel: FPS: 348 FrameTime: 2.878 ms
[bump] bump-render=high-poly: FPS: 389 FrameTime: 2.573 ms
[bump] bump-render=normals: FPS: 530 FrameTime: 1.888 ms
[bump] bump-render=height: FPS: 467 FrameTime: 2.144 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 161 FrameTime: 6.230 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 67 FrameTime: 15.045 ms
[pulsar] light=false:quads=5:texture=false: FPS: 374 FrameTime: 2.674 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 55 FrameTime: 18.430 ms
[desktop] effect=shadow:windows=4: FPS: 175 FrameTime: 5.716 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 240 FrameTime: 4.184 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 239 FrameTime: 4.201 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 259 FrameTime: 3.875 ms
[ideas] speed=duration: FPS: 238 FrameTime: 4.216 ms
[jellyfish] <default>: FPS: 139 FrameTime: 7.245 ms
[terrain] <default>: FPS: 19 FrameTime: 52.984 ms
[shadow] <default>: FPS: 171 FrameTime: 5.869 ms
[refract] <default>: FPS: 40 FrameTime: 25.279 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 323 FrameTime: 3.098 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 294 FrameTime: 3.412 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 324 FrameTime: 3.091 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 322 FrameTime: 3.115 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 258 FrameTime: 3.886 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 322 FrameTime: 3.108 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 322 FrameTime: 3.111 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 258 FrameTime: 3.884 ms
=======================================================
                                  glmark2 Score: 306 
=======================================================
```

With `mesa-panfrost-git-24.1.0-devel-git-4ab1904496`:
```
=======================================================
    glmark2 2023.01
=======================================================
    OpenGL Information
    GL_VENDOR:      Mesa
    GL_RENDERER:    Mali-G610 (Panfrost)
    GL_VERSION:     OpenGL ES 3.1 Mesa 24.1.0-devel (git-4ab1904496)
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=24 stencil=0 samples=0
    Surface Size:   3840x2160 windowed
=======================================================
[build] use-vbo=false: FPS: 480 FrameTime: 2.086 ms
[build] use-vbo=true: FPS: 578 FrameTime: 1.732 ms
[texture] texture-filter=nearest: FPS: 468 FrameTime: 2.140 ms
[texture] texture-filter=linear: FPS: 469 FrameTime: 2.134 ms
[... 8 lines cut ...]
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 159 FrameTime: 6.306 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 67 FrameTime: 15.106 ms
[pulsar] light=false:quads=5:texture=false: FPS: 370 FrameTime: 2.704 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 63 FrameTime: 15.900 ms
[desktop] effect=shadow:windows=4: FPS: 177 FrameTime: 5.663 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 239 FrameTime: 4.193 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 237 FrameTime: 4.226 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 258 FrameTime: 3.883 ms
[ideas] speed=duration: FPS: 226 FrameTime: 4.444 ms
[jellyfish] <default>: FPS: 137 FrameTime: 7.315 ms
[terrain] <default>: FPS: 14 FrameTime: 76.404 ms
[shadow] <default>: FPS: 178 FrameTime: 5.646 ms
[refract] <default>: FPS: 43 FrameTime: 23.365 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 320 FrameTime: 3.125 ms
[conditionals] fragment-steps=5:vertex-steps=0:^[[1;5D FPS: 291 FrameTime: 3.440 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 321 FrameTime: 3.123 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 320 FrameTime: 3.126 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 254 FrameTime: 3.941 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 321 FrameTime: 3.120 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 320 FrameTime: 3.127 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 255 FrameTime: 3.927 ms
=======================================================
                                  glmark2 Score: 304 
=======================================================
```

7Ji, 2024-08-12:
https://github.com/7Ji-PKGBUILDs/.meta/issues/49

Closing

closed 2024-08-12 by 7Ji