7Ji/orangepi5-archlinuxarm #54: not booting after sudo pacman -Syu and kernel update

insidesources, 2024-09-18:
Hello,

I am downloading this image: https://github.com/7Ji/orangepi5-archlinuxarm/releases/download/nightly/ArchLinuxARM-aarch64-OrangePi5-20240731_021110-rkloader-orangepi_5_plus.img.gz and putting it on my orange pi 5 plus.

After booting and running sudo pacman -Syu and reboot, it doesn't boot again (there is a kernel update when i did pacman) have you or anyone experienced this?

I could just not update it, but then in the future i dont want to black screen this if theres an issue with updating the kernel and packages

GutsyHeart, 2024-09-20:
I have the same thing.

7Ji, 2024-09-20:
I don't have time yet to dig on this. But before I have time to sort this out, you can add the kernel package(s) to `IgnorePkg` option in `/etc/pacman.conf`, or try other kernels by modifying the `DEFAULT` line in `/boot/extlinux/extlinux.conf`

JoseGalRe, 2024-09-21:
i have same issue too

7Ji, 2024-09-22:
I found the problem. Currently the builder would remove `/etc/booster.yaml` after building. And booster, after when the config is missing, would refuse to build new initrd. And as pacman hook `60-booster-remove.hook` would remove the old initrd before updating the kernel package, `90-booster-install.hook` would create new initrd after installing/updating new kernel, this results in missing initrd after a kernel update. 

I would fix the builder to include an empty `booster.yaml`, and new images should come online soon

7Ji, 2024-09-22:
Fixed in https://github.com/7Ji/orangepi5-archlinuxarm/commit/52a452e0a292ed2f4b3763298638cfbca0af7942 , new images are tested on opi 5 and they work OK.

If you already have an installation and do not want to re-flash the image, simply `sudo touch /etc/booster.yaml` would solve the problem.

closed 2024-09-22 by 7Ji