7Ji/orangepi5-archlinuxarm #55: Orange Pi Max Support

fleeco, 2024-10-02:
Heya - first and foremost, thanks for the great work!

I'm looking at your rk_bootloader repository and I see that you support the OPI Max.  Any chance you could add that over to this repo?  If not, I'm goign to take a stab at building the bootloader and using the rootfs image here - but...I have a feeling what takes me hours might take you minutes!

7Ji, 2024-10-04:
I see you've created a fork and did some modification and it has created an image for 5 max. That should suit your need.

For why I didn't introduce more boards to this repo though, :
1. As I write the builder script with local-first in mind (so it could be used without a Github Actions runner), everything is done in a single, huge and long step seen by the Github Actions runner. That is, during the life time of a single action step, all profiles would need to be run and all different images shall be created in the same single run. However, as the disk space of Github Actions runner is very limited, this breaks when I enable too many profiles. You could verify this by checking the recent commits where I have to disable some kernels and profiles to make the runner working again.
2. I always insist on verifying the stuffs I write on actual hardware before pushing them online. I only have 5 and 5 plus, and don't have max or pro. For rkloader I added them anyway as users who dare to burn the bootloader would know enough about the booting process to not brick the board and complain in issues, but I don't want to risk many coming here and complaining about non-working images once I added "support" for 5 max/pro without actual testing.

I'm working on another project https://github.com/7Ji/aimager that would function as an atomic cross-architrecture  Arch and Arch-derived image builder, and it shall be possible to build by multiple Github Actions "step"s. And people shall be able to either come to submit their support feature PR or fork the project easily. However I'm really busy in real life and had only got some free time to spend the China National Day holiday with my family.

codabrink, 2025-03-09:
Appreciate you making a max build, @fleeco